Ultimaker/Cura #4540: Feature Request: Show Settings Different From Profile

sleemanj, 2018-10-10:
In simple terms, a manner in which we can select to see thew settings which are different from the profile (the ones with the circular arrow), and ONLY those.  Rather than having to show all setting and then scroll through the list looking for circular arrows.

[image]

fieldOfView, 2018-10-11:
At one point in the history of the SettingVisibilityPresets menu, I was planning to add this (https://github.com/Ultimaker/Cura/commit/503c612509346d5a215953255d9230c1e2dc9288#diff-47c5e3ec9d6202c11682ef44841784ac), but I did not get around to implementing it.

ChrisTerBeke, 2018-10-11:
Note that you can see a diff already when you switch to another profile and you have unsaved changes.

Ghostkeeper, 2018-12-14:
A couple of UX things to think about:
* If you don't have any changes, what will it display?
* If the value for a profile setting is currently at 0.4mm and you want to type "0.45mm", should the setting briefly disappear after displaying "0." and before displaying "0.45"?
* If you click on one of the reset buttons, should the setting disappear too?

Ghostkeeper, 2018-12-14:
Ah, hold on. This is a duplicate of #3713. I'll post those concerns there.

closed 2018-12-14 by Ghostkeeper