Ultimaker/Cura #4541: [3.4.1] Objects in imported photos are stuck in a rectangle of surrounding material, even when transparency is set

rallant92, 2018-10-11:
**Application Version**
3.4.1

**Platform**
Windows 10 64bit

**Printer**
None

**Steps to Reproduce**
1. Start Cura
2. Import a picture that is a two color silhouette, like a logo.  A great example is the gear silhouettes at https://www.deviantart.com/dinn/art/Gears-JPG-189806582.  Drag the image to Cura.
3. Fix the import to be "Darker is higher".
4. Select the OK button in the dialog to import.

**Actual Results**
Notice all gears are joined together, on top of a 1mm high rectangle.

**Expected results**
The gears should all be separate.

**Additional Information**
I tried to work around by switching the photo to a PNG with transparency so there literally is no background.  Same problem.

I also tried setting the min height to 0.  There still seems to be the same connecting material appearing, so no go.  However, slicing and switching to layer view, finally reveals that the connecting material is not printed, making this as a workaround.  This is really not obvious because of the solid view appearance.

IMO the default "Lightest is highest" does not match almost any of the artwork available.  Just Google Image search for gear silhouette or logo silhouette.  Logo ideas does have at least a few dark backgrounds, but not even half.

If the dark color is not black the import model will never reach the height target.  I was given an image using dark blue that caused some surprise. Maybe models should be scaled to always reach the bounds specified, because specifying the bounds makes them expected?

Ghostkeeper, 2018-12-13:
The "Lightest is highest" option tends to work well for height maps and for photos. Height maps tend to be brighter in places where they are higher, and photos tend to have bright objects in the foreground.

Ghostkeeper, 2018-12-13:
PNG transparency is never loaded. Cura doesn't support that right now.

Most people who print pictures or height maps and a lot of cases where you want to print text or something need a background to keep it all together. You don't want the rivers in your height map to actually break your model in two.

The people that don't can just set the "Base" to 0mm. Then you won't get the background. This is expected behaviour. It's just not the default.

closed 2018-12-13 by Ghostkeeper