Ultimaker/Cura #4544: Dual Extrusion Build Plate Volume

Leifcald, 2018-10-11:
I Cant print this sprocket with support layer using dual extruders although the second extruder is no way near the boundries for the second extruder.

Shouldnt the diffrent extruders slice diffrently to allow maximum build volume.?

**Application Version**
3.5.0

**Platform**
Windows 7 X64

**Printer**
Ultimaker 3 Extended

[image]

[image]

diegopradogesto, 2019-09-11:
In a Ultimaker 3 there is no room to move the right extruder all the way to the left of the buildplate, or the left one all the way to the right. The new Ultimaker S5 already solves that.

We cannot fix this in Cura, it's hardware dependent.

closed 2019-09-11 by diegopradogesto

Ghostkeeper, 2019-09-12:
The report was about the right extruder not needing to go all the way to the left of the build plate, since the part that was printed by that extruder doesn't go that far towards the edge. In the picture it's only the support that's being printed by the other extruder, which is more towards the centre of the build plate.