Ultimaker/Cura #4545: [Feat.REQ] Set speed for printing on top of supports/sup.interface

swatti87, 2018-10-11:
See the red and green with no infill. Those were run at full speed on top of the supports. 
Dat be baaad.

[image]

They wont really connect to the supports or the support-interface so when the nozzle changes direction in the end-corner, it pulls the whole string of wall with it. It evens out at later layers but its ugly.

Notice that part of the layer is on supports, while some parts are not.

If the area printing on the supports had speed setting, OR the entire layer went at set speed, it would be great.

At the moment i cant really raise the supports any higher or they tend to fuse with the print.

swatti87, 2018-10-11:
Tested "bridging options" and they dont effect the walls... 

"Everything on top of supports considored bridge" -option would work.

alekseisasin, 2018-10-12:
Can you share the project?

Ghostkeeper, 2018-12-18:
"Bridge Wall Speed" should affect the walls. It does, or at least when I try it.

I think that is sufficient for this use case. We should not implement a second feature to do the same.

closed 2018-12-18 by Ghostkeeper