Ultimaker/Cura #4546: Cura 3.5.0 doesn't start

PHvdWa, 2018-10-11:
after shutting down and new start of Windows. I'm using the Preview Build  18252 and the Release 17164.345 too. On both computers Cura 3.5.0 doesn't start.
The Application only could be seen as a background process.
Cura 3.4.1 on both computers installed starts without any problem.
It is the same whether Cura 3.4.1 is installed or not.
Even installing 3.5 as administrator doesn't help.
As you could see is Cura_3.5 only as background process und Cura 3.4.1 as Background and Foreground Process available. 

[image]

Another problem. After having deinstalled Cura 3.5.0 you can't install it another time. Message see Screenshot.

[image]

Any ideas?

Peter

LipuFei, 2018-10-11:
This seems to be the same problem as https://github.com/Ultimaker/Cura/issues/4546

LipuFei, 2018-10-11:
@Ghostkeeper That error message mentioning "Arcus", I have a sneaky suspicion that it has something to do with that dead lock problem. Maybe using a Windows 10 Preview Build is the most reliable way to reproduce this problem.

PHvdWa, 2018-10-11:
It's the same problem with the release version of Windows 10 1803 not 1809.

alekseisasin, 2018-10-12:
what do you mean? Windows 10 **1803** not **1809**
Do you want to say after windows update it stopped to work?

PHvdWa, 2018-10-12:
No, 1809 isn't installed yet.

tonywmnix, 2018-10-13:
I see this  as well. If I use Windows 8 compatibility mode it works but much slower. I also tried deleting the cura folder in AppData and have mixed success having it start natively.

alekseisasin, 2018-10-15:
This is "Windows 8 compatibility mode it works but much slower" really bad because it related to Windows configuration and I am not sure we can provide a good solution.

Ghostkeeper, 2018-12-21:
Does this still occur in 3.6 or 4.0? We've changed some things with how the communication to the slicing back-end works that could affect this bug.

PHvdWa, 2018-12-21:
Hello,
no, meanwhile Cura starts normal.

Peter

closed 2018-12-21 by Ghostkeeper

tonywmnix, 2018-12-21:
I tried 4.0 yesterday and saw this problem. For 3.6 I have been able to use it without using the slow windows compatibility mode by first logging out of my company VPN which requires a proxy setting and then repeatedly starting cura and killing it when it doesn't start. After 5-10 times it comes up.
Thanks