Ultimaker/Cura #4547: Cura 3.5 displays "unable to slice" immediately after loading & displaying STL

Inspironator, 2018-10-11:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Inspironator, 2018-10-11:
Application Version  3.5.0

Platform HP Desktop running Win10 - 64 bit OS

Printer  Not Connected, I use a micro-USB, but it is a Creality CR-10

Steps to Reproduce   Install Cura 3.5.0 and enabled it to read all available file types during installation; open the base.stl file on the desktop from within Cura's File commands.

Actual Results:  The object drops onto the build plate, then in a flash, Cura displays "unable to slice" and shows 0 for print time and cost spec info.

Expected results:  Expected it to display "ready to slice" along with print time, cost spec info.

Additional Information:  I have run 3.4.1 and 3.5.0 side by side simultaneously, and after uninstalling 3.4.1 and the results are the same with 3.5.0.  I searched for the local/cura log file and can find that file.  I did find a log file in the appdata-roaming- cura, whose contents are:

2018-10-11 11:53:15,205 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin FileLogger
2018-10-11 11:53:15,283 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin FirmwareUpdateChecker
2018-10-11 11:53:15,312 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeGzReader
2018-10-11 11:53:15,334 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeGzWriter
2018-10-11 11:53:15,351 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeProfileReader
2018-10-11 11:53:15,391 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeReader
2018-10-11 11:53:15,538 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeWriter
2018-10-11 11:53:15,578 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin ImageReader
2018-10-11 11:53:15,597 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin LegacyProfileReader
2018-10-11 11:53:15,621 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin LocalContainerProvider
2018-10-11 11:53:15,644 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin LocalFileOutputDevice
2018-10-11 11:53:15,666 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin MachineSettingsAction
2018-10-11 11:53:15,686 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin ModelChecker
2018-10-11 11:53:15,707 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin MonitorStage
2018-10-11 11:53:15,729 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin PerObjectSettingsTool
2018-10-11 11:53:15,750 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin PostProcessingPlugin
2018-10-11 11:53:15,769 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin PrepareStage
2018-10-11 11:53:15,794 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin RemovableDriveOutputDevice
2018-10-11 11:53:15,826 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SimulationView
2018-10-11 11:53:15,851 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SliceInfoPlugin
2018-10-11 11:53:15,874 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SolidView
2018-10-11 11:53:15,897 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SupportEraser
2018-10-11 11:53:15,937 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin Toolbox
2018-10-11 11:53:15,966 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin CameraTool
2018-10-11 11:53:15,994 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin MirrorTool
2018-10-11 11:53:16,022 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin RotateTool
2018-10-11 11:53:16,051 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin ScaleTool
2018-10-11 11:53:16,078 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SelectionTool
2018-10-11 11:53:16,105 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin TranslateTool
2018-10-11 11:53:16,145 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UFPWriter
2018-10-11 11:53:16,177 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UltimakerMachineActions
2018-10-11 11:53:16,295 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UM3NetworkPrinting
2018-10-11 11:53:16,327 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UpdateChecker
2018-10-11 11:53:16,332 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
2018-10-11 11:53:16,423 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin USBPrinting
2018-10-11 11:53:16,447 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UserAgreement
2018-10-11 11:53:16,483 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade21to22
2018-10-11 11:53:16,510 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade22to24
2018-10-11 11:53:16,533 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade25to26
2018-10-11 11:53:16,556 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade26to27
2018-10-11 11:53:16,576 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade27to30
[... cut ...]
2018-10-11 11:53:49,103 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\UM\Scene\Iterator\Iterator.py", line 16, in __init__
2018-10-11 11:53:49,106 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 107, in _fillStack
2018-10-11 11:53:49,109 - ERROR - [Thread-4] UM.Logger.logException [88]: ZeroDivisionError: division by zero
2018-10-11 11:54:07,319 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

Inspironator, 2018-10-12:
Yup, first ERROR in my log file as well as ARROWTRON is  "ERROR - [MainThread] UM.Logger.logException [84]: Exception: Exception calling decoration getConvexHullHead: object of type 'NoneType' has no len()"

I think this is not an installation problem.  
I have no clue as to the cause except a software error.

Ghostkeeper, 2018-12-19:
I think this depends on the model that you're using. I'm not getting the same results when following your reproduce steps. Problems with the convex hull are usually happening when you have a model that has pieces of 0 width.

closed 2026-09-06 by GregValiant