Ultimaker/Cura #4550: [3.5] Unable to add custom materials

TakuroFukamizu, 2018-10-11:
**Application Version**
3.5.0

**Platform**
Windows 10 Pro 1803

**Printer**
Custom Printer: Dobot Mooz-2

**Steps to Reproduce**
1. Change `Compatible material diameter` to `1.75 mm`  at Machine Settings dialog.
2. Open the `Materials` preference dialog
3. Duplicate the `Generic PLA` material and change `Diameter` to `1.75 mm`.

**Actual Results**
- Showing an information dialog that says "The new filament diameter is set to 1.75mm., which is not compatible with the current extruder. Do you wish to continue?".
- When I press the YES button, that custom material was disappeared.

**Expected results**
- Shouldn't show the above information dialog because filament diameter is compatible.
- Don't dissolve current material setting.

surfrock66, 2018-11-03:
This is happening on Cura 3.5.1 on Linux (the Appimage) as well.

closed 2018-11-08 by diegopradogesto

surfrock66, 2018-11-15:
FYI this is NOT fixed in Cura 3.6; exact same thing is still happening

surfrock66, 2018-11-15:
Additionally, looking at Issue 4582, in Cura 3.6.0 the "Printer Settings" menu is not possible in the sidebar in the Linux AppImage.  Even when "All Settings" are checked, that sub-menu is not an option.  I don't see a place to configure a compatible filament diameter in the machine or extruder setting; this may be a Linux AppImage issue?

fieldOfView, 2018-11-16:
Could you test this prerelease version of th "Printer Settings" plugin?
http://files.fieldofview.com/cura/PrinterSettingsPlugin-v5.0.0-2018-11-16T14_28_49Z.curapackage
Just drop the file into the Cura window, and restart Cura.

surfrock66, 2018-11-17:
I tested that, changed the "Diameter" to 1.75, the setting stuck.  Went to settings, changed the material diameter to 1.75, it popped up the error and when I clicked continue the material profile disappeared.  The bug persists.  3.6.0 AppImage on Debian Buster.

fieldOfView, 2018-11-18:
In the materials dropdown in the sidebar, do you see Ultimaker materials or a whole list of brands?

surfrock66, 2018-11-18:
In settings, or in the actual print setup?  I don't see anything in the print setup sidebar, but here's what I have in the settings:

Materials Sidebar: https://imgur.com/ZCYZ81R
Print Settings Sidebar: https://imgur.com/wrwWmPm
Materials Settings: https://imgur.com/b5F0zWz

fieldOfView, 2018-11-18:
Ah, this is interesting. What printer definition are you using? It seems to disable material selection.

surfrock66, 2018-11-18:
This is a profile for the CraftBot Plus which I imported AGES ago and has been brought forward as I've updated.

GUI Printer Settings: https://imgur.com/1GrSEP9
GUI Extruder Settings: https://imgur.com/dJDo7Vz

Contents of ~/.local/share/cura/3.6/extruders/fdmextruder+%232.extruder.cfg
```
[general]
version = 4
name = Extruder
id = fdmextruder #2

[metadata]
machine = CraftBot+
type = extruder_train
enabled = True
setting_version = 5
position = 0

[containers]
0 = fdmextruder #2_user
1 = fdmextruder_#2_cb+.pla.2mm.205
2 = normal
3 = generic_pla #5
4 = empty_variant
5 = fdmextruder #2_settings #2
6 = fdmextruder
```
Contents of ~/.local/share/cura/3.6/CraftBot%2B.global.cfg
```
[general]
version = 4
name = CraftBot+
id = CraftBot+

[metadata]
type = machine
setting_version = 5

[containers]
0 = CraftBot+_user
1 = craftbot_cb+.pla.2mm.205
2 = normal
3 = empty_material
4 = empty_variant
5 = CraftBot+_settings
6 = craftbot
```
Contents of ~/.local/share/cura/3.6/definitions/craftbot.def.json
```
{
    "id": "craftbotplus",
    "version": 2,
    "name": "CraftBot+",
    "inherits": "fdmprinter",
    "metadata": {
        "visible": true,
        "author": "wlldm",
        "manufacturer": "CraftUnique",
        "category": "CraftBot",
        "file_formats": "text/x-gcode",
        "icon": "icon_ultimaker2",
        "platform_texture": "logo_craft-cura.png",
		"platform": "craftbot_platform.obj",
        "platform_offset": [0, 0, 0],
        "has_materials": false,
		"weight": 3
		
    },

    "overrides": {
        "machine_name": { "default_value": "CraftBot+" },
        "machine_heated_bed": {
            "default_value": true
        },
        "machine_width": {
            "default_value": 250
        },
        "machine_height": {
            "default_value": 200
        },
        "machine_depth": {
            "default_value": 200
        },
        "machine_center_is_zero": {
            "default_value": false
        },
        "machine_nozzle_size": {
            "default_value": 0.4
        },
        "material_diameter": {
            "default_value": 1.75
        },
        "machine_nozzle_heat_up_speed": {
            "default_value": 2
        },
        "machine_nozzle_cool_down_speed": {
            "default_value": 2
        },
		"wall_thickness": {
			"default_value": 1.0, "visible": false
		},
		"top_bottom_thickness": {
			"default_value": 1.0, "visible": false
		},
		"bottom_thickness": {
			"default_value": 1.0, "visible": false
		},
		"material_print_temperature": {
			"default_value": 220, "visible": true
		},
		"material_bed_temperature": {
			"default_value": 0, "visible": true
		},
		"speed_print": {
			"default_value": 50.0
		},
		"speed_infill": {
			"default_value":80.0, "visible": true
		},
		"speed_wall": {
			"default_value": 35.0, "visible": true
		},
		"speed_topbottom": {
			"default_value": 35.0, "visible": true
		},
		"speed_travel": {
			"default_value": 100.0
		},
		"speed_layer_0": {
			"default_value": 20.0, "visible": false
		},
		"retraction_speed": {
			"default_value": 30.0, "visible": false
		},
		"retraction_amount": {
			"default_value": 0.6, "visible": false
		},
		"retraction_hop": {
			"default_value": 0.0, "visible": false
		},
		"support_enabel": {
			"default_value": false
		},
        "machine_head_polygon": {
            "default_value": [
                [-75, -18],
                [-75, 35],
                [18, 35],
                [18, -18]
            ]
        },
		"layer_height": {
            "default_value": 0.2
        },
		"layer_height_0": {
            "default_value": 0.2
        },
		"shell_thickness":{
			"default_value": 1.0
		},
        "gantry_height": {
            "default_value": 55
        },
        "machine_gcode_flavor": {
            "default_value": "RepRap (Marlin/Sprinter)"
        },
        "machine_start_gcode": {            
			"default_value": "; -- START GCODE -- \nM4002 P1 ;set heat program 1 \nM1203 A88 D66 F1200 ;set higher accel and decel \nG28 X0 Y0 ;home XY \nG28 Z0 ;home Z\nG91 ;set relative movements \nG0 F100 ;speed 100mm/sec \nG0 E20 ;extrude 20mm of filament\nG90 ;absolute movements \n; -- end of START GCODE --" 
        },
        "machine_end_gcode": {
			"default_value": "; -- END GCODE -- \nM104 S0 ;turn off hotend \nM140 S0 ;turn off bed\nG91 ;set to relative positioning \nG1 E-20 F300 ;retract filament to release pressure \nG90 ;set to absolute positioning \nG0 X0 Y200 ;move the head to the left at the back \nM84 ;turn off steppers \n; -- end of END GCODE --"
        }
    }
}
```

fieldOfView, 2018-11-18:
In .../definitions/craftbot.def.json, it makes sense to change the has_materials line (since you obviously want to use materials):
```"has_materials": true,```
In .../extruders/fdmextruder+%232.extruder.cfg, under containers, change the line starting with 3:
```3 = generic_pla_175```

surfrock66, 2018-11-18:
I made those changes but the behavior has not changed; same popup and everything and then my duplicated material profile disappears.

fieldOfView, 2018-11-18:
Could you zip up the entire ~/.local/share/cura/3.6 folder? Something is telling your configuration that your printer needs 3mm filament (instead of the 1.75mm you want), and I can keep guessing, or I can just have a look at the entire configuration.

surfrock66, 2018-11-18:
https://www.surfrock66.com/3.6.tar.gz

fieldOfView, 2018-11-18:
I had a look. The easiest way to get things working is to go into Machine Settings again. Now that you have changed the has_material property, there's a material diameter on the Extruder 1 tab. Set that to 1.75.

The better way would be to add an extruder definition, which includes the material diameter. See this commit, which did that for all single extruder printers in Cura:
https://github.com/Ultimaker/Cura/commit/aba3795d0b51b3ca8fa5307bd380a0a8fade6aff

surfrock66, 2018-11-18:
That did it, the compatible filament diameter setting wasn't there before and now it is.  Changing that to 1.75 made all those "disappeared" materials re-appear.

fieldOfView, 2018-11-18:
It wasn't there because the "has_materials" setting was set to False. The real bug here is that the Materials pane of the preferences should not show or let you create or import materials if the printer does not support materials.

diegopradogesto, 2018-11-19:
The list of the materials would depend on the `approximate material diameter`, that is different from the printers that use a 2.85mm filament and 1.75mm filament. That being said, when your active printer is one that uses 2.85mm filament, only the materials with this diameter will show up in the list. If your active printer has the compatible diameter as 1.75mm, then only those materials show up.

Then, when creating a 2.85mm material, if you change it to 1.75mm, you will see a popup indicating that the material is not compatible with the current printer. It doesn't mean that it disappears, it means that it only will show up in the other list.

I don't know if this is your issue @surfrock66 , but if this is the case, it is not a bug, it's how it works.

But indeed I agree with @fieldOfView that it shouldn't be possible at least to activate a material if the printer definitions has the `has_materials` flag set to False. I made a PR for that (#4836)

surfrock66, 2018-11-19:
I didn't see anything in my config that specified an extruder material compatibility at all, is there some sort of default?  The stuff in "fdmextruder+%232.extruder.cfg" wasn't written by me, it seems auto-generated?

surfrock66, 2018-11-19:
Right, but given that, something in Cura assumes 2.85mm material.  Even if the printer definition doesn't specify anything about material or extruders, that 2.85mm material seems (to me, a user that doesn't know the source) the default material.  As someone just above a printing novice, that seems abnormal because my experience shows 1.75 to be much more common.

I would think it would be helpful to add something to the message that might help someone find where in a machine config the sticking point is, or maybe allow for some sort of global material or extruder default separate from a user defined config?

fieldOfView, 2018-11-19:
If you want to know where the default is coming from, it is here:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmextruder.def.json#L246

As far as your other suggestion goes, I remain with my opinion that if your printer definition says that your printer does not support materials, that Cura should not be showing materials. In your case, your machine definition was incomplete (because it lacks an extruder definition of its own, which would have overridden the material_diameter setting).

surfrock66, 2018-11-19:
It makes sense that my profile was incomplete; my profile was pulled off a forum for my printer ages ago and the Craftbot has its own slicer (CraftWare) so Cura profiles were hard to find.  I didn't even know what the "materials=false" thing was.  I just don't know what I would have done to find it without posting here, especially since it was working fine in versions up until 3.4.1.

diegopradogesto, 2018-11-20:
There was an issue previous to 3.4 that we fix by adding extruder definitions to all printer definitions (https://github.com/Ultimaker/Cura/pull/3914). Of course we add this file to all the definitions that we have internally in Cura, but we can't do it for the profiles that you can download somewhere. If you have this issue, I recommend you to create an extruder definition, defining the material diameter and the nozzle size.

provistapp, 2019-02-21:
Add new custom (copy existing) material with settings 1.75 filament and use that material with new printer