Ultimaker/Cura #4551: bad extrusion amount

jmerc77, 2018-10-11:
**Application Version**
3.5.0

**Platform**
windows 10

**Printer**
xyz 2.0a with repetier v0.92 mod
[curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2470530/curaproject.3mf.zip)

**Steps to Reproduce**
set up "custom" printer:
0.4 nozzle
1.75mm filament
150x200x200 (xyz) volume
import parts and slice
attempt to print.

**Actual Results**
prints with half the extrusion. checked material length against my other slicing software and it calculates to about half.

**Expected results**
extrusions should match closely and not half.

smartavionics, 2018-10-11:
I sliced your project and looked at the gcode in S3D and CraftWare and neither show it looking under-extruded. I do wonder though about your settings. You are using a 0.3mm line width with a 0.4mm nozzle and are printing quite fast. Does that combination produce good extrusions?

[image]

jmerc77, 2018-10-12:
i do not know what I did differently, but works now.

closed 2018-10-12 by jmerc77