Ultimaker/Cura #4552: Model generates incorrect gcode

mqudsi, 2018-10-11:
**Application Version**
Cura 3.4.1

**Platform**
Windows 10 64-bit (likely N/A)

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Prepare and print the attached project.

**Actual Results**
I aborted the print once it became apparent that it did not match what Cura was showing me, but midway through the model (where the bulge is in the design) the hollow interior was closed off with a bridge, and the printer began printing a solid form (without a hollow interior). Additionally, about the same place, instead of the exterior being a single surface, it created an exterior ring that was a separate component not attached to the former.

**Expected results**
A single, hollow shape should have been printed.

**Additional Information**
I imagine the STL I imported into Cura has a gap, I'm not trying to say that Cura should have been able to handle it (if that's the case) but merely that even if the STL were bad, whatever Cura rendered on-screen should have been what was (attempted) to print.

Project: [UM3E_32mm Adapter v1.curaproject.zip](https://github.com/Ultimaker/Cura/files/2470609/UM3E_32mm.Adapter.v1.curaproject.zip)

What Cura shows:

[image]

Aborted print gone wrong:

smartavionics, 2018-10-11:
>I imagine the STL I imported into Cura has a gap, I'm not trying to say that Cura should have been able to handle it (if that's the case) but merely that even if the STL were bad, whatever Cura rendered on-screen should have been what was (attempted) to print.

You mean like this image showing the Layer View ?

[image]

mqudsi, 2018-10-11:
Well never mind then!

In my defense, my layer view is broken:

[image]

As you can see, there are no controls for actually controlling which layer is viewed. Perhaps this is fixed in 3.5, I should upgrade and see if I need to  file a bug report.

closed 2018-10-11 by mqudsi

smartavionics, 2018-10-11:
You need to slice the object to get the layer view controls and the layers displayed. Bye!

fieldOfView, 2018-10-12:
Automatic slicing was turned off a couple of versions ago. You can turn it back in the General prrferences.

mqudsi, 2018-10-12:
I think it might be helpful for a message to be displayed where the controls would be when the model isn’t sliced?