Ultimaker/Cura #4556: aplicacion se cierra al cambiar de máquina

angelotti85, 2018-10-12:
La aplicación se cierra automáticamente cuando intento cambiar de perfil de máquina.
Recien he actualizado a v3.5.

¿Alguien tiene una solución?

¡ Gracias !

diegopradogesto, 2018-10-16:
Por favor, rellena completamente la infomación que se pide, sino no podremos ayudarte. Además te pediría que lo hagas en inglés, así más gente podrá entender el problema y quizá ayudarte a encontrar una solución. También otro usuario con el mismo problema puede seguir la conversación.
----
Please, fill out the information that we ask for, otherwise we can't help you. Besides I will ask you to do it in English so more people can understand the problem and maybe help you find a solution. Also another user with the same problem will be able to follow the conversation.

diegopradogesto, 2018-11-08:
Closing, not enough info provided.

closed 2018-11-08 by diegopradogesto