Ultimaker/Cura #4558: [3.5.0] Unable to slice in "One at a time" sequence if object placed at edges.

JannSchulte, 2018-10-12:
**Application Version**
3.5.0

**Platform**
Windows 10 64bit 1803

**Printer**
Tevo Tornado (with CR10 profile)

**Steps to Reproduce**
add small objects (1x1x1) at edges of print area
change Print Sequence to "one at a time"

**Actual Results**
Error: Unable to slice 

**Expected results**
working slicing, since "All at once" slices successfully and the objects are pretty small and printable.

**Additional Information**
[image]
[image]

Ghostkeeper, 2018-12-19:
It can't print it because it's going to print a brim around each one of these.

closed 2018-12-19 by Ghostkeeper

dimake, 2019-03-07:
Bit of help for One at a Time settings on Cura 3.4.1 which is unable to slice
[image]

Ghostkeeper, 2019-03-08:
Maybe all of your meshes are too high to fit in the build volume? Some printer models reduce the build volume height when slicing multiple models in one-at-a-time mode because the suspension of the print head can collide with the models.