Ultimaker/Cura #4559: [3.5.0

smartavionics, 2018-10-12:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

smartavionics, 2018-11-01:
Master branch (presumably, 3.6 also) still gets this horribly wrong.

With optimizePaths() being called...

[image]

Without...

[image]

smartavionics, 2018-11-01:
This bug is printer agnostic. Whether it occurs or not appears to be controlled by the area being filled, line widths, etc.

[UMS5_tip_led_case_bottom.3mf.zip](https://github.com/Ultimaker/Cura/files/2539553/UMS5_tip_led_case_bottom.3mf.zip)

[image]

Ghostkeeper, 2018-12-19:
We now consider this issue fixed (in 4.0 beta 2). Some refactor work is coming on there, but this bug should no longer occur now.

closed 2018-12-19 by Ghostkeeper

ctmakro, 2019-01-10:
[image]

so when do we get 4.0 beta 2?