Ultimaker/Cura #4562: [3.5.0] Z Hop doesn't move down again

PierreMonico, 2018-10-12:
**Application Version**
3.5.0

**Platform**
mac OS High Sierra

**Printer**
Cutom Printer (Anet A8 with RAMPS)

**Steps to Reproduce**
Slice model with a profile I tweaked for a few weeks and has Z Hop enabled. 
Level bed.
Start print.

**Actual Results**
Skirt is perfect.
Head Z Hops 0.5 as expected but starts the first layer of object in this position, doesn't go back down to 0.2.
The first layer is unable to stick.

**Expected results**
The first layer should be printed at the same height as the skirt. 

**Additional Information**
My Z Hop is set at 0.5. Layer is at 0.2 for both skirt and model.
I checked the gcode and indeed, after the skirt I get a:
G1 F300 Z0.7
G0 F2133.3 X63.981 Y64.75 Z0.7
Followed by a:
G1 F300 Z0.2
So it should work. But I can actually see that the head is way to high after printing the skirt. It somehow stays at 0.7. I don't know much about gcode so I can't tell if the above should work.

Whenever I disable Z Hop everything works fine. 

Please find attached:
* A model with which it happens (it does it for all models)
* The generated gcode
* My profile
* The project file

[cura_bug.zip](https://github.com/Ultimaker/Cura/files/2473737/cura_bug.zip)

I can also include pictures where you can clearly see it happen if needed.

Thank you for your help and for making Cura.

smartavionics, 2018-10-12:
Hi. Thanks for the report and the files. I have looked at the gcode and I cannot see any problem in the first layer. The image below shows the nozzle back down at 0.2mm just before it starts the first wall after the skirt. Could it be a mechanical problem with the printer? Also, does it still have a problem if you don't change the accel/jerk for travel moves?

[image]

PierreMonico, 2018-10-18:
Sorry I couldn't access my printer for a week.
What do you mean by not changing the accel/jerk? Like using a default Cura profile? 

Strangely enough, I reflashed Marlin, activated the Z Hop again and everything works now. Probably a Marlin issue or something mechanic (most probably).

Thanks again.

closed 2018-12-24 by Ghostkeeper