Ultimaker/Cura #4563: Exporting and Importing profiles doesn't copy extruder settings.

pynej, 2018-10-13:
**Application Version**
3.5
**Platform**
OSX
**Printer**
N/A
**Steps to Reproduce**
Create profile, set setting, import and export, check settings.

**Actual Results**
Some settings are back at default, those that seam to be tied to a specific extruder. In my case I have not touched any settings related to multiple extruders as I only have the one.
**Expected results**
All the settings to be replicated
**Additional Information**
[image]
[image]

diegopradogesto, 2018-11-08:
Sorry for the delay. Can you please test the same with the version 3.6-BETA that is available in our website? If it doesn't work, could you please send the 3.5 configuration folder to us? (You can find it in `$USER/Library/Application Support/cura/`

nallath, 2019-03-01:
Closing due to inactivity: Aditional information was requested but not provided

closed 2019-03-01 by nallath