Ultimaker/Cura #4564: Cura 3.5.0 crashing on changing Printer type from drop down option as well as Configure Cura Printers options

cdeox, 2018-10-13:
**Application Version**
<Ultimaker Cura 3.5.0 >

**Platform**
<Windows 10 Pro 64-bit>

**Printer**
<Creality CR 10S custom profile>

**Steps to Reproduce**
<1. Open Cura 3.5.0
        2. Select any other Printer profile from top right corner.
        3. Alternatively, go to Preferences - Configure Cura - Printer Profiles - Activate any Printer profile.>

**Actual Results**
< Cura application crashes >

**Expected results**
< Printer profile change to selected profile >

**Additional Information**
<Happens irrespective any models have been opened or not. Happens to all profiles.
[CR10S_personal.curaproject.curaproject.zip](https://github.com/Ultimaker/Cura/files/2475144/CR10S_personal.curaproject.curaproject.zip)
>

fieldOfView, 2018-10-13:
Please add a link to the Cura.log file

cdeox, 2018-10-13:
Attaching the Cura GUI log file and the DxDiag file:
[DxDiag.zip](https://github.com/Ultimaker/Cura/files/2475176/DxDiag.zip)

[cura.zip](https://github.com/Ultimaker/Cura/files/2475179/cura.zip)

ubershy, 2018-10-14:
I have the same issue with two printer profiles ("AnetA6_0.4" and "AnetA6_0.8" in cura.log). Can't switch to another printer profile because of crash when I try to do so.
Switching printer profiles worked some time ago in Cura 3.5. Maybe after I updated/created some print settings profiles the configuration became corrupted.
Here is the log:
[cura.zip](https://github.com/Ultimaker/Cura/files/2476224/cura.zip)

ArtieH, 2018-10-14:
Same issue when switching between two custom printers.  Crashes using drop down and also when using the manage printers activate button.

ubershy, 2018-10-15:
@ArtieH thank you for providing workaround

nallath, 2018-10-16:
Although I wasn't able to reproduce it, I do think that I squashed your bug (6eb8b754903f3fe1db5baa55ad90305fda94de38)

ArtieH, 2018-10-17:
The bug is still kicking in version 3.5.1.

The workaround still clears the error temporarily.

nallath, 2018-10-17:
That's because I fixed it in master, so it should be resolved for 3.6

closed 2018-12-21 by Ghostkeeper