Ultimaker/Cura #4565: Cura 3.5.0 - Mouse pointer is offest in settings but not main program.

Grievousbee, 2018-10-13:
**Application Version**
3.5.0

**Platform**
Win10 Professional x64

**Printer**
Doesn't matter / Irrelevant

**Steps to Reproduce**
For me everytime its opened.

**Comments**
Cura 3.4.1 was fine but now on 3.5.0 my mouse pointer is offset in the settings window but not in the main program.
However on my laptop which is also Win10 x64 Home everything is fine.
The pc this happens on uses an Nvidia 970GTX card with AMD 8320 (I think) cpu whereas the laptop is an AMD A8 APU with inbuilt graphics.

I did update graphics card drivers but no difference.
Program is still useable but its tricky changing settings. I dont want to have to use laptop all thje timne as that is far too slow.
Is there anything else I need to update?

I did search for this problem but the only thing I could find mentioned was a few versions ago and related to Intel gfx drivers so that was of no use to me.

Chris

Ghostkeeper, 2018-12-19:
We haven't seen this ourselves yet. It's really hard to determine what's causing it or how to do anything about it.

So that issue you found about the Intel graphical drivers was a different GPU, but maybe it could help for you to update your graphical drivers as well?

Grievousbee, 2018-12-20:
Hi,

Thanks for the reply.

I am keeping graphics drivers updated buit no difference so far.

I now have another PC I am using and this is fine.
It runs Linux Mint with and AMD cpu that has on board graphics but its the earlier version not the A series, however as this is a lot slower than my main one I'd rather not use it for most of my slicing

It seems only to be this nVidia card tha is the problem however I dont have any other nVidia based graphics to try out on.
As I say, 3.4.1 works fine and as this does all I need I'll stick with this versio9n on the problem pc for now.

What I will do is boot to the Linux install on the problem PC and see if it still has the problem and if not its almost certainly a Windows driver problem.

Will report back on the outcome.

Chris

Grievousbee, 2018-12-20:
**Found the problem!**

It is in fact an incompatibility with WindowBlinds10!

Have now told WindowBlinds to ignore Cura.exe and everything works as it should.

Strange that 3.4.1 worked fine even with Windowblinds though, but hey who cares, it works now. :-)

I will now close this down.

Chris

closed 2018-12-20 by Grievousbee