Ultimaker/Cura #4566: [master] CuraEngine takes 100% CPU before first slice

smartavionics, 2018-10-13:
**Application Version**

master

**Platform**

Linux (Ubuntu 16.04.5)
**Printer**

Kossel XL

**Steps to Reproduce**

Start cura.

**Actual Results**

Cura loads as normal but when CuraEngine is started it consumes 100% CPU until the first slice has been done. Once the first slice has been done the CuraEngine process goes away.

**Expected results**

On starting cura, the CuraEngine process should not consume 100% CPU before the first slice.

**Additional Information**

After cura has loaded and before the first slice, ps reports this:
````
~/Downloads$ ps aux | grep cura
markb    11614  0.0  0.0   4504   784 pts/2    S    14:48   0:00 /bin/sh /usr/local/bin/cura.sh
markb    11621 32.9  2.0 2067452 338176 pts/2  Sl   14:48   0:28 python3 /opt/cura/bin/cura
markb    11643 99.0  0.0 119564  5716 pts/2    RNl  14:48   1:22 /opt/cura/bin/CuraEngine connect 127.0.0.1:49674
````

After the first slice, ps reports this:
````
~/Downloads$ ps aux | grep cura
markb    11614  0.0  0.0   4504   784 pts/2    S    14:48   0:00 /bin/sh /usr/local/bin/cura.sh
markb    11621 35.9  2.4 2143048 405324 pts/2  Sl   14:48   1:01 python3 /opt/cura/bin/cura
````

[cura.log.zip](https://github.com/Ultimaker/Cura/files/2475523/cura.log.zip)

diegopradogesto, 2018-10-17:
@smartavionics we did some changes also in libArcus. Did you compile it again the last days? That should fix the issue. If not, let us know and we will investigate.

smartavionics, 2018-10-17:
Yes, I am using the latest version.

diegopradogesto, 2018-10-17:
Ok, let's investigate then. Devs.: CURA-5831

alekseisasin, 2018-10-17:
I could reproduce this issue by doing manipulation with Cura and CuraEngine but never had for the normal Cura start. @smartavionics can you make some changes in CuraEngine to see an extra log
In Application.cpp
Line 225:
    while (communication->hasSlice())
    {
        // Can you add some output, like std::cout<<"Check flag"<< std::flush<<std::endl;
        communication->sliceNext();
    }
My assumption this is the place where CuraEngine has infinite loop for the fist Cura run, if it is so then we will see alot of message.

smartavionics, 2018-10-17:
Yes, I did that and can confirm that it is wizzing round that loop continuously when Cura is first started. After the first slice, the output stops but then while doing subsequent slices, output occurs again but stops when the slice has completed. Hope this helps.

Vandrasc, 2018-10-26:
Hi @smartavionics , when time allows you could you please confirm the fix? Looks good on our side!

Many thanks!

smartavionics, 2018-10-26:
Hi, I have given it a quick test and it appears to be OK. Engine no longer using 100% CPU on start and sliced a couple of times with no problems.

diegopradogesto, 2018-11-08:
This is fixed in 3.6.

closed 2018-11-08 by diegopradogesto

smartavionics, 2018-12-14:
> This is fixed in 3.6.

Perhaps, but it has now become unfixed again in the master branch and it behaves as before. CuraEngine grinds away at 100% CPU until the first slice has been done.

smartavionics, 2018-12-14:
No, please ignore that last comment from me, I had an old libArcus.a hanging about which got used instead of the .so. Sorry!