Ultimaker/Cura #4567: Cura 2.5.0 - Extruder Temperature incorrectly written to GCode

MacLemon, 2018-10-13:
**Application Version**

Cure 2.5.0

**Platform**

macOS 10.12.6 (Sierra)

**Printer**

Prusa i3MK2

**Steps to Reproduce**

Slice an Object with ABS, Set Extruder temperature to 250°C, export GCODE.

**Actual Results**

Temperature in GCODE is set to 210°C
```
M104 S210 ; set extruder temp
M140 S100 ; set bed temp
M190 S100 ; wait for bed temp
M109 S210 ; wait for extruder temp
```
Cooldown Section is correctly set to 0°C.

**Expected results**

Temperature should correspond to what has been set by the user.

**Additional Information**

This is a regression over 3.4.1.

stwoodward, 2018-10-14:
I can confirm this same issue.  Initial print temperature is always set to **210** in the GCODE no matter what temperature is entered in the Cura settings for: 

- **"Printing Temperature Initial Layer"**
- **"Initial Printing Temperature"**

Temperature after initial layer does appear to be updating correctly when the GCODE is generated, it is only the initial layer print temperature that appears to have an issue with updating in the GCODE.

Note this is for **Cura 3.5.0**

**Application Version**
&nbsp;  &nbsp; &nbsp; Cure 3.5.0

**Platform**
&nbsp;  &nbsp; &nbsp; Windows 10 v1803 OS buidl 17134.345

**Printer**
&nbsp;  &nbsp; &nbsp; Creality Ender 3

Vandrasc, 2018-10-15:
The fix will be added in the next release, thank you both for your feedback! Devs fixed in CURA-5793.

closed 2018-10-15 by Vandrasc