Ultimaker/Cura #4568: **Application Version**  3.5.0, "UNABLE TO SLICE"

Inspironator, 2018-10-13:
**Platform** HP Desktop running Win10 - 64 bit OS

**Printer**  Not Connected, I use a micro-USB, but it is a Creality CR-10

**Steps to Reproduce**   Install Cura 3.5.0 and enabled it to read all available file types during installation; open the base.stl file on the desktop from within Cura's File commands.

**Actual Results:**  The object drops onto the build plate, then in a flash, Cura displays "unable to slice" and shows 0 for print time and cost spec info.

**Expected results:**  Expected it to display "ready to slice" along with print time, cost spec info.

**Additional Information:**  I have run 3.4.1 and 3.5.0 side by side simultaneously, and after uninstalling 3.4.1 and the results are the same with 3.5.0.  I searched for the local/cura log file and can find that file.  I did find a log file in the appdata-roaming- cura, whose contents are:

2018-10-11 11:53:15,205 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin FileLogger
2018-10-11 11:53:15,283 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin FirmwareUpdateChecker
2018-10-11 11:53:15,312 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeGzReader
2018-10-11 11:53:15,334 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeGzWriter
2018-10-11 11:53:15,351 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeProfileReader
2018-10-11 11:53:15,391 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeReader
2018-10-11 11:53:15,538 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin GCodeWriter
2018-10-11 11:53:15,578 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin ImageReader
2018-10-11 11:53:15,597 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin LegacyProfileReader
2018-10-11 11:53:15,621 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin LocalContainerProvider
2018-10-11 11:53:15,644 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin LocalFileOutputDevice
2018-10-11 11:53:15,666 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin MachineSettingsAction
2018-10-11 11:53:15,686 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin ModelChecker
2018-10-11 11:53:15,707 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin MonitorStage
2018-10-11 11:53:15,729 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin PerObjectSettingsTool
2018-10-11 11:53:15,750 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin PostProcessingPlugin
2018-10-11 11:53:15,769 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin PrepareStage
2018-10-11 11:53:15,794 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin RemovableDriveOutputDevice
2018-10-11 11:53:15,826 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SimulationView
2018-10-11 11:53:15,851 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SliceInfoPlugin
2018-10-11 11:53:15,874 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SolidView
2018-10-11 11:53:15,897 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SupportEraser
2018-10-11 11:53:15,937 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin Toolbox
2018-10-11 11:53:15,966 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin CameraTool
2018-10-11 11:53:15,994 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin MirrorTool
2018-10-11 11:53:16,022 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin RotateTool
2018-10-11 11:53:16,051 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin ScaleTool
2018-10-11 11:53:16,078 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SelectionTool
2018-10-11 11:53:16,105 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin TranslateTool
2018-10-11 11:53:16,145 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UFPWriter
2018-10-11 11:53:16,177 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UltimakerMachineActions
2018-10-11 11:53:16,295 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UM3NetworkPrinting
2018-10-11 11:53:16,327 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UpdateChecker
2018-10-11 11:53:16,332 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
2018-10-11 11:53:16,423 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin USBPrinting
2018-10-11 11:53:16,447 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin UserAgreement
2018-10-11 11:53:16,483 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade21to22
2018-10-11 11:53:16,510 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade22to24
2018-10-11 11:53:16,533 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade25to26
2018-10-11 11:53:16,556 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade26to27
2018-10-11 11:53:16,576 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade27to30
2018-10-11 11:53:16,599 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade30to31
2018-10-11 11:53:16,632 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade32to33
2018-10-11 11:53:16,662 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade33to34
2018-10-11 11:53:16,690 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin VersionUpgrade34to35
2018-10-11 11:53:16,718 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin SimpleView
2018-10-11 11:53:16,756 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin X3DReader
2018-10-11 11:53:16,797 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin XmlMaterialProfile
2018-10-11 11:53:16,822 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [421]: Loaded plugin XRayView
2018-10-11 11:53:16,846 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [401]: Plugin BarbarianPlugin uses an incompatible API version, ignoring
2018-10-11 11:53:16,875 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [401]: Plugin OrientationPlugin uses an incompatible API version, ignoring
2018-10-11 11:53:16,897 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [128]: Looking for old configuration files to upgrade.
2018-10-11 11:53:16,901 - DEBUG - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [278]: Config file [.\Creality+CR-10+%232.global.cfg] is of version [4000004], which is different from the defined version [2000000], no upgrade task for it from type [definition_changes].
2018-10-11 11:53:16,906 - DEBUG - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [278]: Config file [.\Creality+CR-10+%232_settings.inst.cfg] is of version [4000004], which is different from the defined version [2000001], no upgrade task for it from type [definition_changes].
2018-10-11 11:53:16,911 - DEBUG - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [278]: Config file [.\Creality+CR-10+fdmextruder_settings.inst.cfg] is of version [4000004], which is different from the defined version [2000001], no upgrade task for it from type [definition_changes].
2018-10-11 11:53:16,915 - DEBUG - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [278]: Config file [.\Creality+CR-10+%232_settings.inst.cfg] is of version [4000004], which is different from the defined version [2000002], no upgrade task for it from type [definition_changes].
2018-10-11 11:53:16,920 - DEBUG - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [278]: Config file [.\Creality+CR-10+fdmextruder_settings.inst.cfg] is of version [4000004], which is different from the defined version [2000002], no upgrade task for it from type [definition_changes].
2018-10-11 11:53:16,925 - DEBUG - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [278]: Config file [.\Creality+CR-10+%232_settings.inst.cfg] is of version [4000004], which is different from the defined version [2000003], no upgrade task for it from type [... cut ...]
2018-10-11 11:53:49,106 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\OneAtATimeIterator.py", line 107, in _fillStack
2018-10-11 11:53:49,109 - ERROR - [Thread-4] UM.Logger.logException [88]: ZeroDivisionError: division by zero
2018-10-11 11:54:07,319 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

_Originally posted by @Inspironator in https://github.com/Ultimaker/Cura/issues/4547#issuecomment-429089612_

Inspironator, 2018-10-13:
Arrowtron also has this same issue, and he posted it yesterday as a comment to my Original post #4547.  His log file is also listed in the original post.

rickgode, 2018-10-14:
I have the same issue.  Windows 10 64bit just installed Cura 3.5 this morning and Unable to Slice.  Cura 3.4.1 is installed as well and slices just fine.

rickgode, 2018-10-14:
I copied the last bit of my cura.log files  Doesn't mean much to me except the error "division by zero"  Maybe that's bad...

2018-10-14 10:51:54,810 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.5/resources/qml/Cura.qml:211: TypeError: Property 'setExtruderIndex' of object MachineManager(0x1f9b2419520) is not a function
2018-10-14 10:52:01,998 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2018-10-14 10:52:02,062 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2018-10-14 10:52:02,149 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.5/resources/qml/DiscardOrKeepProfileChangesDialog.qml:81: TypeError: Cannot read property of null
2018-10-14 10:52:11,075 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [294]: Attempting to kill the engine process
2018-10-14 10:52:11,075 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [300]: Killing engine process
2018-10-14 10:52:11,080 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [303]: Engine process is killed. Received return code 1
2018-10-14 10:52:11,085 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [218]: Previous socket existed. Closing that first.
2018-10-14 10:52:11,137 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [178]: Socket state changed to Listening
2018-10-14 10:52:11,168 - INFO - [MainThread] UM.Backend.Backend.startEngine [74]: Started engine process: C:\Program Files\Ultimaker Cura 3.5\CuraEngine.exe
2018-10-14 10:52:11,184 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [94]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 3.5\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Ultimaker Cura 3.5\\resources\\definitions\\fdmprinter.def.json', '']
2018-10-14 10:52:11,283 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] 
2018-10-14 10:52:11,287 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] Cura_SteamEngine version 3.5.0
2018-10-14 10:52:11,290 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] Copyright (C) 2018 Ultimaker
2018-10-14 10:52:11,294 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] 
2018-10-14 10:52:11,297 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] This program is free software: you can redistribute it and/or modify
2018-10-14 10:52:11,300 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] it under the terms of the GNU Affero General Public License as published by
2018-10-14 10:52:11,302 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] the Free Software Foundation, either version 3 of the License, or
2018-10-14 10:52:11,305 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] (at your option) any later version.
2018-10-14 10:52:11,307 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] 
2018-10-14 10:52:11,310 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] This program is distributed in the hope that it will be useful,
2018-10-14 10:52:11,312 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2018-10-14 10:52:11,315 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2018-10-14 10:52:11,317 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] GNU Affero General Public License for more details.
2018-10-14 10:52:11,320 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] 
2018-10-14 10:52:11,322 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] You should have received a copy of the GNU Affero General Public License
2018-10-14 10:52:11,338 - DEBUG - [Thread-17] UM.Backend.Backend._backendLog [94]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2018-10-14 10:52:11,340 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-10-14 10:52:11,343 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [182]: Socket state changed to Connected
2018-10-14 10:52:11,345 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [172]: Backend connected on port 49674
2018-10-14 10:52:11,456 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Exception calling decoration getConvexHullHead: object of type 'NoneType' has no len()
2018-10-14 10:52:11,463 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2018-10-14 10:52:11,466 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\UM\Scene\SceneNode.py", line 242, in callDecoration
2018-10-14 10:52:11,468 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\Scene\ConvexHullDecorator.py", line 96, in getConvexHullHead
2018-10-14 10:52:11,471 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\3.5\build\inst\lib\python3.5\site-packages\cura\Scene\ConvexHullDecorator.py", line 243, in _compute2DConvexHeadMin [... cut ...]
2018-10-14 10:53:11,807 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2018-10-14 10:55:04,850 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.5\resources\shaders\toolhandle.shader]...
2018-10-14 10:55:13,062 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2018-10-14 10:59:04,382 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

diegopradogesto, 2018-10-19:
Can you please provide a project file? (Project > Save...)

smartavionics, 2018-10-21:
OK, I think this is maybe the same problem or perhaps a variation on the theme. The project file (https://community.ultimaker.com/applications/core/interface/file/attachment.php?id=61293) attached to the forum thread https://community.ultimaker.com/topic/24956-cura-351-unable-to-slice/ causes these messages to appear which could be relevant...

````
2018-10-21 21:54:12,530 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Exception calling decoration getConvexHullHead: could not convert string to float: '[[0,0],[0,0],[0,0],[0,null]]'
2018-10-21 21:54:12,531 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2018-10-21 21:54:12,531 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/UM/Scene/SceneNode.py", line 242, in callDecoration
2018-10-21 21:54:12,531 - ERROR - [MainThread] UM.Logger.logException [88]:     return getattr(decorator, function)(*args, **kwargs)
2018-10-21 21:54:12,531 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/cura/Scene/ConvexHullDecorator.py", line 114, in getConvexHullHead
2018-10-21 21:54:12,532 - ERROR - [MainThread] UM.Logger.logException [88]:     head_with_fans = self._compute2DConvexHeadMin()
2018-10-21 21:54:12,532 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/cura/Scene/ConvexHullDecorator.py", line 264, in _compute2DConvexHeadMin
2018-10-21 21:54:12,532 - ERROR - [MainThread] UM.Logger.logException [88]:     head_and_fans = self._getHeadAndFans()
2018-10-21 21:54:12,532 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/cura/Scene/ConvexHullDecorator.py", line 254, in _getHeadAndFans
2018-10-21 21:54:12,532 - ERROR - [MainThread] UM.Logger.logException [88]:     return Polygon(numpy.array(self._global_stack.getHeadAndFansCoordinates(), numpy.float32))
2018-10-21 21:54:12,533 - ERROR - [MainThread] UM.Logger.logException [88]: ValueError: could not convert string to float: '[[0,0],[0,0],[0,0],[0,null]]'
````

And also
````
2018-10-21 22:06:54,841 - ERROR - [Thread-3] UM.Logger.logException [84]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x9743a88> caused an exception
2018-10-21 22:06:54,841 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2018-10-21 22:06:54,841 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/UM/JobQueue.py", line 121, in run
2018-10-21 22:06:54,842 - ERROR - [Thread-3] UM.Logger.logException [88]:     job.run()
2018-10-21 22:06:54,842 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [178]: Socket state changed to Listening
2018-10-21 22:06:54,854 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 163, in run
2018-10-21 22:06:54,855 - ERROR - [Thread-3] UM.Logger.logException [88]:     for node in OneAtATimeIterator(self._scene.getRoot()): #type: ignore #Ignore type error because iter() should get called automatically by Python syntax.
2018-10-21 22:06:54,855 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/cura/OneAtATimeIterator.py", line 41, in __init__
2018-10-21 22:06:54,856 - ERROR - [Thread-3] UM.Logger.logException [88]:     super().__init__(scene_node)  # Call super to make multiple inheritance work.
2018-10-21 22:06:54,856 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/UM/Scene/Iterator/Iterator.py", line 16, in __init__
2018-10-21 22:06:54,856 - ERROR - [Thread-3] UM.Logger.logException [88]:     self._fillStack()
2018-10-21 22:06:54,857 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/cura/OneAtATimeIterator.py", line 106, in _fillStack
2018-10-21 22:06:54,857 - ERROR - [Thread-3] UM.Logger.logException [88]:     min_coord = self.getMachineNearestCornerToExtruder(self._global_stack)
2018-10-21 22:06:54,857 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/opt/cura/lib/python3/dist-packages/cura/OneAtATimeIterator.py", line 59, in getMachineNearestCornerToExtruder
2018-10-21 22:06:54,857 - ERROR - [Thread-3] UM.Logger.logException [88]:     x = coord[0] - extruder_offsets[0]
2018-10-21 22:06:54,858 - ERROR - [Thread-3] UM.Logger.logException [88]: TypeError: unsupported operand type(s) for -: 'str' and 'int'

````

rickgode, 2018-10-22:
I think there is a bug that causes it not to slice if the Print Sequence is set to "One at a Time"  Setting it to "All at Once" at first says unable to slice, but then it slices just fine.

[MLG.zip](https://github.com/Ultimaker/Cura/files/2499963/MLG.zip)

Ricky

smartavionics, 2018-10-22:
Looking at this some more, the project file contains anet\ A8_settings.inst.cfg which looks like this:
````
[general]
version = 4
name = anet A8_settings
definition = custom

[metadata]
type = definition_changes
setting_version = 5

[values]
extruders_enabled_count = 1
gantry_height = 0
machine_depth = 220
machine_end_gcode = M104 S0 ;extruder heater off
        M140 S0 ;heated bed heater off
        M106 S0 ; turn off fan
        G92 E1
        G1 E-1 F300 ;Retract the filament
        G28 X0 Y0
        M84
machine_head_with_fans_polygon = [[0,0],[0,0],[0,0],[0,null]]
machine_heated_bed = True
machine_height = 240
machine_start_gcode = ;Sliced at: {day} {date} {time}
        G21 ;metric values
        G90 ;absolute positioning
        M82 ;set extruder to absolute mode
        M107 ;start with the fan off
        G28 ;home all axis
        G1 Z15.0 F{travel_speed} ;move the platform down 15mm
        G92 E0 ;zero the extruded length
        G1 F200 E3 ;extrude 10mm of feed stock
        G92 E0 ;zero the extruded length again
        G1 F{travel_speed}
        M117 Printing...
machine_width = 220

````

How did `machine_head_with_fans_polygon` get that value? Is it a bad printer def or something the user has entered or has Cura dreamed it up?

smartavionics, 2018-10-22:
The anet A8_settings.inst.cfg file that is in the project that gives the above errors contains this:
````
machine_head_with_fans_polygon = [[0,0],[0,0],[0,0],[0,null]]
````
Is that a user error or did Cura create the null?

rickgode, 2018-10-22:
I have no idea how the machine got that value.  Attached is a pic of my Printer Settings.  Also it doesn't seem to be the "correct" start and end gcode?  When I'm back home, I'll try and save the .3mf file from Cura 3.4.1.  

[image]

rickgode, 2018-10-22:
Attached is a project file from Cura 3.4.1 

I don't know where in Cura I would change "machine_head_with_fans_polygon"?

[3.4.1upper_filament_guide_new_matter_mod-t.curaproject.zip](https://github.com/Ultimaker/Cura/files/2501887/3.4.1upper_filament_guide_new_matter_mod-t.curaproject.zip)

diegopradogesto, 2018-10-24:
We changed the way how the disabled space between the objects when printing one-at-a-time mode is calculated. In this case because your values are all zeros, it is unable to slice. As a workaround you can indicate the values for the dimensions of the printhead (rather than zero).

fieldOfView, 2018-10-24:
The X min, Y min, X max and Y max fields in machine settings together form the machine_head_with_fans_polygon setting (making the polygon a rectangle). Having all these at 0 makes the polygon a rectangle with 0 width and 0 height.

Inspironator, 2018-10-24:
Changing the Print Sequence to "all at once" did enable Cura 3.5.0 to slice for me also.   I changed the printhead settings and I'll report back the results for the new machine settings.

rickgode, 2018-10-27:
Thanks everyone!  I measured my printhead settings and put them in.  Everything looks good.  I made my Gantry height 9999mm so I could get some taller models to print "one at a time" I just ensure the placement of the models on the build plate won't run into the print head.  It sliced great!

https://www.youtube.com/watch?v=6YZLyJAGDMQ

Thanks!

Ricky

danielassr, 2018-11-14:
Hello!

I wonder if anyone can help me, please!

After the last windows update (a month ago), I have been experiencing problems with cura.  This one keeps presenting the "unable to slice" when I upload a file. I uninstalled and installed a few times, the 3.5.1 version and the 3.4.1 (the last one that worked), but this problem remained. Today, I installed the last version (3.6) to check if this problem had disappeared, but unfortunately no.

So, my cura logfile is the following:

2018-11-14 17:30:16,396 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2018-11-14 17:30:16,427 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdateChecker
2018-11-14 17:30:16,443 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdater
2018-11-14 17:30:16,457 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzReader
2018-11-14 17:30:16,477 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzWriter
2018-11-14 17:30:16,493 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeProfileReader
2018-11-14 17:30:16,524 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeReader
2018-11-14 17:30:16,544 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeWriter
2018-11-14 17:30:16,563 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ImageReader
2018-11-14 17:30:16,573 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LegacyProfileReader
2018-11-14 17:30:16,598 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2018-11-14 17:30:16,617 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2018-11-14 17:30:16,634 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MachineSettingsAction
2018-11-14 17:30:16,653 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ModelChecker
2018-11-14 17:30:16,667 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MonitorStage
2018-11-14 17:30:16,687 - WARNING - [MainThread] UM.Preferences.getValue [97]: Tried to get the value of non-existing setting cura/active_mode.
2018-11-14 17:30:16,694 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PerObjectSettingsTool
2018-11-14 17:30:16,713 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PostProcessingPlugin
2018-11-14 17:30:16,723 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PrepareStage
2018-11-14 17:30:16,759 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RemovableDriveOutputDevice
2018-11-14 17:30:16,793 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimulationView
2018-11-14 17:30:16,813 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SliceInfoPlugin
2018-11-14 17:30:16,837 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SolidView
2018-11-14 17:30:16,857 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SupportEraser
2018-11-14 17:30:16,893 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin Toolbox
2018-11-14 17:30:16,916 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2018-11-14 17:30:16,944 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2018-11-14 17:30:16,966 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2018-11-14 17:30:16,997 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2018-11-14 17:30:17,017 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2018-11-14 17:30:17,046 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2018-11-14 17:30:17,079 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UFPWriter
2018-11-14 17:30:17,103 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UltimakerMachineActions
2018-11-14 17:30:17,234 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UM3NetworkPrinting
2018-11-14 17:30:17,264 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2018-11-14 17:30:17,274 - INFO - [Thread-3] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2018-11-14 17:30:17,323 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin USBPrinting
2018-11-14 17:30:17,344 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UserAgreement
2018-11-14 17:30:17,377 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade21to22
2018-11-14 17:30:17,397 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade22to24
2018-11-14 17:30:17,424 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade25to26
2018-11-14 17:30:17,448 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade26to27
2018-11-14 17:30:17,473 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade27to30
2018-11-14 17:30:17,497 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade30to31
[... cut ...]
2018-11-14 17:32:02,205 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [809]: Backend quit with return code 0. Resetting process and socket.

I must mention that the file which I tried to slice was sliced and printed many times, in older versions of cura (3.4 and earlier).
I have the windows 10 and a 64 bits CPU.

Thank you all in advance.

I would really appreciate any help :)

P.S. I keep thinking that could be a problem with a deleted drive by the windows update, but please let me know your suggestions.

smartavionics, 2018-11-15:
Thanks for the file, unfortunately, that isn't the project file, it only contains the model. Please do File -> Save.

danielassr, 2018-11-15:
Oh sorry, the right file is this one:

[UM3_Assembly 3.zip](https://github.com/Ultimaker/Cura/files/2585091/UM3_Assembly.3.zip)

smartavionics, 2018-11-15:
Thanks for that. Well, I loaded the project into my version of Cura (which is a bit like 3.6.0) and it sliced OK. Maybe there has been a recent fix? is it worth you trying 3.6?

[image]

danielassr, 2018-11-15:
The only thing that happened was the windows update. After that, whatever the version, the unable to slice was always appearing. Now I have the 3.6 version. I can't imagine what the problem might be since I previously sliced and printed that.

danielassr, 2018-11-15:
This is what it is presented in my Cura

[image]

smartavionics, 2018-11-15:
Ah, that screenshot shows that it is actually in the process of slicing. Perhaps the slicing engine has crashed. Can you find cura.log (Help -> Show Configuration Folder), zip it up and attach. Thanks.

danielassr, 2018-11-15:
Hmm, I hope is because of that. As requested, the file:

[cura.zip](https://github.com/Ultimaker/Cura/files/2585590/cura.zip)

smartavionics, 2018-11-15:
Your log contains lines like this:
````
2018-11-15 11:46:35,577 - DEBUG - [Thread-37] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Could not connect to the given address
2018-11-15 11:46:35,655 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [809]: Backend quit with return code 0. Resetting process and socket
.
````

It appears to be having a problem communicating with the back end. Do you have a firewall active? If so, does it work OK if you disable the firewall?

danielassr, 2018-11-15:
Even with the firewall disabled, the problem remains and it does not work :|

diegopradogesto, 2018-11-19:
For you, @danielassr the problem seems to be the firewall. I think it's the exact same issue as #4223 or #3334 or #3118. All of them related with the libArcus port is not open. Please take a look at the proposed solutions.