Ultimaker/Cura #4570: Cura 3.5.0 wrongly combines two models (one above the other)

Cyberman-tM, 2018-10-14:
As the title says - I've a cube I want to print which has indentions on each side (for later inlays), since I know the filament will sag on the bottom side, I wanted to try to put a little plate underneath, so it can't sag all the way down.

For some reason Cura thinks the two clearly separate models are one.

**Application Version**

3.5.0

**Platform**

Windows 10

**Printer**

Wanhao i3

**Steps to Reproduce**

Load STL, slice, check in layer view

**Actual Results**

The first bottom layer of the cube clearly shows that the plate below is considered to reach INTO the cube.

**Expected results**

The two models should be printed "separately" but at the same time.

Attachments
[image]
[image]

fieldOfView, 2018-10-14:
Make sure there is some vertical distance between the main model and your support. Or - better yet - select the small block, and use "Per Model Settings" to change the mesh type to "Print as Support".

Cyberman-tM, 2018-10-14:
> Make sure there is some vertical distance between the main model and your support.
There is - about 0.4mm - you can see it in the Tinkercad screenshot.

>Or - better yet - select the small block, and use "Per Model Settings" to change the mesh type to "Print as Support".
Holy cow, I didn't even know this existed! Cool!
I've printed it with full Cura support now (worked quite well, my printer isn't adjusted fine enough, but that's not something Cura can help with, I think?). I'll keep that setting in mind, though - definitely something that can be useful.

diegopradogesto, 2018-10-16:
> Holy cow, I didn't even know this existed! Cool!

XDDDDDDD

This is a duplicate of #4317 . We already change the default behavior of the "Ignore Small Z Gaps" setting to be false, but it's still on master, so it will work by default on next release. In the meanwhile you can manually disable the setting and it should work.

closed 2018-10-16 by diegopradogesto