Ultimaker/Cura #4571: Cura 3.5.0 no Infillprint

Revo75, 2018-10-14:
**Application Version**
Cura 3.5.0

**Platform**
Windows 7 64bit

**Printer**
Custom CoreXY

**Steps to Reproduce**
in all Models

**Actual Results**
no print Infill, in the Settings 70%

**Expected results**
anyone models

**Additional Information**
can't upload a pic

smartavionics, 2018-10-15:
Please save the project and zip the .3mf file and attach to this issue. Without that, we can do nothing.

Revo75, 2018-10-15:
here

Revo75, 2018-10-15:
next try 
[2te CFFFP_Y Axis Threaded Rod Mount.zip](https://github.com/Ultimaker/Cura/files/2478283/2te.CFFFP_Y.Axis.Threaded.Rod.Mount.zip)

smartavionics, 2018-10-15:
OK, got that. I think the problem is that you have gradual infill steps set to 2. Set it to 0 and you get the density you expect.

Revo75, 2018-10-15:
who is the setting for this `?

smartavionics, 2018-10-15:
[image]

Revo75, 2018-10-15:
filldepth ist here 0.3

smartavionics, 2018-10-15:
> filldepth ist here 0.3

Sorry, I do not understand what you are saying there. Please provide more info or post a screenshot.

Revo75, 2018-10-15:
ok i have recalc and its the same .3mf and a pic is attached
[3te CFFFP_Y Axis Threaded Rod Mount.zip](https://github.com/Ultimaker/Cura/files/2478596/3te.CFFFP_Y.Axis.Threaded.Rod.Mount.zip)
[image]

Revo75, 2018-10-15:
7-18 has no infill partially 19 is in the air

smartavionics, 2018-10-15:
Have you tried setting gradual infill steps to 0?

Revo75, 2018-10-15:
ahh :) thanks, i have this not being set, in 3.4.x no prob with infill

closed 2018-10-15 by Revo75