Ultimaker/Cura #4572: [3.5] Linking line in spiralize mode

Liger0, 2018-10-14:
**Application Version**
3.5

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Use spiralize mode on this model

**Actual Results**
It should print normally

**Expected results**
It prints a random line in the air as a linking point. It is the same thing that happens when trying to print 2 stls, but as I reported it someone closed the ticket saying it doesn't have to be fixed.

**Additional Information**
[image]
[spiral mode.zip](https://github.com/Ultimaker/Cura/files/2476819/spiral.mode.zip)

smartavionics, 2018-10-14:
It is true that Cura currently doesn't handle this model well but I am not convinced that model is suitable for printing as a spiral. The spiralization can cope with smoothly changing outlines but where you have discontinuities like this model has, I don't see how it can be printed as one spiral line. If the z-seam position could be fixed such that it was correct for all the layers from bottom to top then maybe it could be printed but, in general, I don't think that is possible so I would imagine that there is not a general solution that would work for all models of this type.

Even if that diagonal line wasn't there, the model would not be printable as the unsupported edges would droop terribly.

diegopradogesto, 2018-10-16:
Indeed, this is not the intention of the spiralize feature. I don't think this can be handled in a better way.

closed 2018-10-16 by diegopradogesto