Ultimaker/Cura #4573: [3.5] Monoprice Select Mini V1 & V2 Compatible material diameter is incorrect

ProfessorPinky, 2018-10-15:
**Application Version**
3.5

**Platform**
Windows 10 Professional 64bit

**Printer**
Monoprice Select Mini V1 and Monoprice Select Mini V2 (E3D)

**Steps to Reproduce**
Add new printer

**Actual Results**
Compatible material diameter setting is 2.85

**Expected results**
Compatible material diameter setting should be 1.75

**Additional Information**
It is easy enough to manually change the diameter to 1.75 but the reason I am filing and issue is because I have seen post about it on the subreddit and Facebook groups for the Select Mini
[Monoprice Select Mini V2 (E3D).zip](https://github.com/Ultimaker/Cura/files/2477529/Monoprice.Select.Mini.V2.E3D.zip)
[image]
[cura.log.zip](https://github.com/Ultimaker/Cura/files/2477543/cura.log.zip)

Vandrasc, 2018-10-15:
Devs see CURA-5817

LipuFei, 2018-10-15:
Fixed in 73de72094

fieldOfView, 2018-10-15:
Note that this is also wrong for Alya3dp, Creality CR10S4, Creality CR10s5, deltabot, grr_neo, kupido, makeR pegasus, maker_starter and printrbot_play_heated extruders.

Ghostkeeper, 2018-10-15:
What is that deltabot printer actually? I can't find it online.

Ghostkeeper, 2018-10-15:
See this commit to fix the rest: https://github.com/Ultimaker/Cura/commit/02efd7a1f5a312ea4124a45025ac2dc3bdac99da

closed 2018-12-21 by Ghostkeeper