Ultimaker/Cura #4574: Cura often disturbs standard Mac function keys fo luminosity

mimobis, 2018-10-15:
As far as I use Cura (around 3.4.0) I notice that, after a while, function keys for luminosity are no more working .

**Application Version**
? -> 3.4.0 -> 3.5.0.

**Platform**
Mac OS X High Sierra.

**Printer**
Geeetech i3, Dagoma Neva Magis... independent from printer.

**Steps to Reproduce**
Use Cura with common operations: view layers, generate code...

**Actual Results**
The keyboard keys decrease and increase luminosity often stop working after a while.
(located with F1 and F2).

**Expected results**
What one can do!

**Additional Information**
To have the keys back, must reboot!

Ghostkeeper, 2018-12-21:
We're unable to reproduce that here on MacOS 10.14 (Mojave) with Cura 4.0.

mimobis, 2018-12-21:
Hello,
Thank you for your answer, I’m still under High Sierra and the issue happens every day I use Cura,  after a while (not immediately).
Days I don’t use Cura, I don’t face the issue, so I think it is the cause, but…

🎄Bon Noël et Joyeuses Fêtes🤶

Cordialement,
---
Michel Morat

closed 2020-06-20 by no-response[bot]