Ultimaker/Cura #4575: [3.5.0] Cura does random connections from supports - support material intersects model

P3D-1, 2018-10-15:
**Application Version**
3.5.0

**Platform**
W7

**Printer**
Ultimaker S5

**Steps to Reproduce**
Open the project file, slice, enable layer view
[UMS5_Breakaway_Problem.zip](https://github.com/Ultimaker/Cura/files/2478803/UMS5_Breakaway_Problem.zip)

**Actual Results**
Random Breakaway lines intersecting the model, leading to unwanted print results.

**Expected results**
No model intersection

**Additional Information**
[image]

May be fixed with https://github.com/Ultimaker/CuraEngine/pull/843 - however currently I am not able to verify.

BagelOrb, 2018-10-15:
FYI: This bug is unrelated to Connect Polygons.

Perhaps this is something which @smartavionics can shed light into off the top of his head?

BagelOrb, 2018-10-15:
See also https://community.ultimaker.com/topic/24775-breakaway-support-intersects-model-with-cura-350/

smartavionics, 2018-10-15:
This slices OK for me using my master branch based Cura so perhaps the problem has already been fixed upstream for a future release. If you want to try for yourself, you could install one of my Cura releases that are based on the master branch.

https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

[image]

BagelOrb, 2018-10-15:
@Appesteijn We should find out which commit fixed this issue and cherry-pick that commit to the 3.5 branch before 3.5.1

I would categorize this as a major bug.

diegopradogesto, 2018-10-16:
Devs. see CURA-5828

P3D-1, 2018-10-16:
The issue is still there for me in Cura 3.5.1.

smartavionics, 2018-10-16:
Do all of the infill patterns cause this problem? If not, could another pattern be used as a workaround? Does turning off the connecting lines setting make a difference?

P3D-1, 2018-10-16:
> Do all of the infill patterns cause this problem? If not, could another pattern be used as a workaround? Does turning off the connecting lines setting make a difference?

All support patterns available show this problem. Turning off the "connect support lines" option doesn't make a difference.

smartavionics, 2018-10-16:
Oh, well it was worth asking the question.

BagelOrb, 2018-10-16:
I've found the problem together with @Appesteijn 

The support bottoms automatically connect infill polygons independent of any user setting.

`FffGcodeWriter::addSupportBottomsToGCode` line 2461

The bug is still present in 3.5.1 and in master.

The Cura team will discuss what to do with this tomorrow.

Appesteijn, 2018-10-16:
[image]

This solves your problem for me.

BagelOrb, 2018-10-16:
For now: a work-around is to set the Support Floor Pattern to ZigZag or to Lines

smartavionics, 2018-10-16:
it's great that you have a workaround. I wonder why I am not seeing the problem here.

[image]

BagelOrb, 2018-10-16:
Please set the visualization coloring to line type. This is hard to look and understand.

The problem might be easier to see on the standard benchy model.

PS: I'm sorry for introducing this bug in the first place!!! Please forgive me.

smartavionics, 2018-10-16:
[image]

P3D-1, 2018-10-16:
Thanks @BagelOrb for the workaround, this also works for me in 3.5.1. You are forgiven ;)

BagelOrb, 2018-10-16:
I don't know where the bug comes from. It's some sort of memory corruption. I can't explain why you don't reproduce the bug now @smartavionics 

I do know that the bug has been fixed in my PR tho.

P3D-1, 2018-11-13:
Seems to be fixed in Cura 3.6.0. Thanks for your work, everyone!

diegopradogesto, 2018-11-19:
Thanks for the feedback!

closed 2018-11-19 by diegopradogesto