Ultimaker/Cura #4576: [3.5.0] Unnecessary travel moves & retractions on top skin layer when

lokster, 2018-10-15:
**Application Version**
3.5.0

**Platform**
Linux

**Printer**
Custom FDM Printer, but It does not matter

**Steps to Reproduce**
Let's say you want to print an object like this:
[image]

It's a simple flat keychain with letters on top (intented to be printed in dual color with manual swap).
If you examine the top skin layer just below the letters, you will find out it looks like this:
[image]

If you print it like this, there is big chance the flat part of the keychain won't have 100% even lines, because there are holes in the skin layer below the letters, which results in multiple retractions/travel moves, and uneven extrusion lines (which are not that prominent, but still degrade the quality).
And also results in increased print time.
What you can do, is to increase "Skin Expand Distance" to something like 10 or whatever works. This closes the gaps, but results in totally unnecessary travel moves and retractions (and still results in "uneven" parts on the skin layer below the letters):
[image]

The workaround is to set the Top Surface Skin Layers to ZERO. The result looks like this:
[image]
And results in very smooth surface below the letters.
Unfortunately this prevents you from printing the "internal" top/bottom layers faster - you have to print them all with the slower speed you want to use for topmost skin layer. Which is a waste of time...

**Expected results**
If the skin expand distance is big and closes all holes, no retractions and travel moves should be necessary on the flat part - the top surface skin layer should look exactly the same as the "top" layers below (just printed at different speed).

Ghostkeeper, 2018-12-21:
The top surface skin lines are planned separately from the other skin lines. They can have a different line width so then they don't attach to the lines of your normal skin nicely.

A fix for this would have them planned at the same time so that the travel moves are optimised at the same time. It should then find that one line ends in the same location as where the next line starts, and make no travel move.

Ghostkeeper, 2019-01-02:
We've discussed this issue with the team and found that it will probably not do a whole lot for quality and printing speed, so we've decided to defer this. We won't be spending time on this issue any time soon.