Ultimaker/Cura #4577: [3.5] Don't put version number into the install path to ensure update instead of new install

AntoineTurmel, 2018-10-15:
**Application Version**
3.5

**Platform**
Windows 10 64-bits

**Printer**
Does not apply

**Steps to Reproduce**
Install Cura 3.4, then install Cura 3.5

**Actual Results**
It installs Cura in a new folder

**Expected results**
Cura 3.4 is updated to Cura 3.5 without having both version installed.

**Additional Information**
...

AntoineTurmel, 2018-10-15:
@nallath I was not prompt to remove old Cura version when installing 3.5.0 could it be linked to #4111 ?

AntoineTurmel, 2018-10-15:
I tried on my main computer, and same issue, the installer did not asked to remove old Cura versions (running on Windows 10 64-bits, French)

[image]
[image]
[image]
[image]
[image]
[image]

diegopradogesto, 2018-10-16:
I think I know what happens. The installer will ask you to remove Cura 3.4 but not Cura 3.4.1 (for instance). It only asks for removing minor versions, but not revisions.

AntoineTurmel, 2018-10-16:
Ok thanks ! Is it planned to have in-app update manager to avoid download/uninstall/install procedure in the future ?

diegopradogesto, 2018-10-16:
Probably not. At least not soon.

AntoineTurmel, 2018-10-18:
The 3.5.1 proposed me to uninstall 3.5.0 👍