Ultimaker/Cura #4578: Monitor screen jumps around

Multimaker, 2018-10-15:
When in the Monitor scree during printing, the info keeps changing and jumping around making it really difficult to click on camera icon or see print duration time

fieldOfView, 2018-10-15:
I'm guessing you are talking about the display on your printer? What printer?

This issuequeue is for Cura, and Cura does not have much control over what is shown on the display of your printer.

Multimaker, 2018-10-16:
Hi,

I am referring to the 'monitor' option within Cura. The dialog box keeps
expanding and minimising by itself... See screenshot...

(Embedded image moved to file: pic09161.gif)

Mark Norman
Product Design Engineer

Kawasaki Logo

Kawasaki Precision Machinery (UK) Ltd, Ernesettle Lane. Plymouth, Devon PL5
2SA
( Phone: +44 (0) 1752 322339       * Email: mnorman@kpm-uk.co.uk       8
Web: www.kpm-eu.com

The information transmitted is intended only for the person or entity to
which it is addressed and may contain confidential and/or privileged
material. Any review, retransmission, dissemination or other use of, or
taking of any action in reliance upon, this information by persons other
than the intended recipient is prohibited. If you receive this in error,
please contact the sender and delete the material from any computer. The
view and/or opinions of individuals expressed within this document do not
necessarily reflect the views of the company.

            <author@noreply.github.com>
Date:	15/10/2018 16:48

This email has been scanned by the Symantec Email Security.cloud service.
For more information please visit http://www.symanteccloud.com

fieldOfView, 2018-10-16:
Attachments in email responses are removed by github. Please attach them in this thread through the website.

What version of Cura are you using?

Multimaker, 2018-10-16:
Hi, Cura 3.5.0
See image....
[image]

nallath, 2018-10-16:
I still don't quite understand what you mean. Could you make a small video of what is going on?

Multimaker, 2018-10-16:
I only have Jing. This creates 'swf' video files. Can you open these?

Many thanks

Mark Norman
Product Design Engineer

Kawasaki Logo

Kawasaki Precision Machinery (UK) Ltd, Ernesettle Lane. Plymouth, Devon PL5
2SA
( Phone: +44 (0) 1752 322339       * Email: mnorman@kpm-uk.co.uk       8
Web: www.kpm-eu.com

The information transmitted is intended only for the person or entity to
which it is addressed and may contain confidential and/or privileged
material. Any review, retransmission, dissemination or other use of, or
taking of any action in reliance upon, this information by persons other
than the intended recipient is prohibited. If you receive this in error,
please contact the sender and delete the material from any computer. The
view and/or opinions of individuals expressed within this document do not
necessarily reflect the views of the company.

Date:	16/10/2018 10:54

nallath, 2018-10-16:
Sure. A few screenshots might also do the trick (provided that it shows the behavior that's going on)

Multimaker, 2018-10-17:
(See attached file: My Movie.mp4)

Mark Norman
Product Design Engineer

Kawasaki Logo

Kawasaki Precision Machinery (UK) Ltd, Ernesettle Lane. Plymouth, Devon PL5
2SA
( Phone: +44 (0) 1752 322339       * Email: mnorman@kpm-uk.co.uk       8
Web: www.kpm-eu.com

The information transmitted is intended only for the person or entity to
which it is addressed and may contain confidential and/or privileged
material. Any review, retransmission, dissemination or other use of, or
taking of any action in reliance upon, this information by persons other
than the intended recipient is prohibited. If you receive this in error,
please contact the sender and delete the material from any computer. The
view and/or opinions of individuals expressed within this document do not
necessarily reflect the views of the company.

Date:	16/10/2018 12:54

Ghostkeeper, 2018-12-21:
We couldn't reproduce this issue back then. Without being able to reproduce it, it's unlikely that we can fix it.

Since then we've redesigned the monitor page anyway so that most likely changed the design more than enough to work out kinks like these. The chance is small that it'll happen again in 4.0.

closed 2018-12-21 by Ghostkeeper