Ultimaker/Cura #4580: 3.5 Filament Flow only a quarter of what it was on 2.7

JohnRichardReed, 2018-10-15:
**Application Version**
Cura 3.5

**Platform**
Windows 10, Files transferred ti printer vis USB stick

**Printer**
Delta Wasp 2040. Using setting Deltabot for printer type and RepRap for gcode flavour

**Steps to Reproduce**
Successful test piece on version 2.7 weighs 23g and starts with a 6mm circular brim.  The brim contains 15 times 0.4mm lines. The printer extruder rotates 8 times to print the brim.

**Actual Results**
Upgraded to version 3.5 and all settings seem to be copied across from the previous version.  The exception to this are the filament and nozzle diameters which had to be changed from 2.85 to 1.75 and 0.5 to 0.4.  Exactly the same STL was used to generate a 3.5 gcode file and the estimated print time was similar that to the previous version.  This version printed but with much reduced filament and the piece only weighed 6g about a quarter of the previous version.  The initial brim print also used 15 lines but only required just under 2 rotations of the extruder, again about a qurter of what it should be.

**Expected results**
The printing of the brim is simple and the only obvious cause of the problem is that the flow is much less than it should be.  Increasing the Flow setting from 100% to about 400% might work but the maximum allowed in Cura is 150%.  I cannot find any setting that might allow a large increase in flow.

**Additional Information**
If anyone can suggest any settings changes that might increase flow significantly then I would be very grateful.

smartavionics, 2018-10-15:
Please save the project (.3mf) file, zip it up and attach to this issue. Thanks.

JohnRichardReed, 2018-10-15:
[D_Surface Flow Test One 150918_fixed.zip](https://github.com/Ultimaker/Cura/files/2480034/D_Surface.Flow.Test.One.150918_fixed.zip)

JohnRichardReed, 2018-10-15:
3MF file attached.  Trust OK. On learning curve ...

JohnRichardReed, 2018-10-15:
Mark

Apologies.  The 3MF file I attached is different to the one described.  The brim is 8mm with 20 lines.  I was trying some changes to investigate and didn’t completely change back.  I think the problem is the same in the uploaded file.  I can try and reproduce the same settings as described if needed.

Apologies again,

John

DjangoBlok, 2018-10-16:
I have the same problem.
Using a craftbot + printer. Copied the printer profile that worked great in version 3.4.1
In 3.5 the flow is far under needed level.

JohnRichardReed, 2018-10-16:
This problem may be on the 3.5.0 bug list in the notes for 3.5.1, so I upgraded to 3.5.1.  This produced a good test piece with much more flow.  The extruder rotated about 6 times for the brim print compared with under 2 for 3.5.0 but not as much as the 8 for 2.7.  The 3.5.1 piece weighed 15g compared with 6g for 3.5.0 but 2.7 was higher with 23g.  About 75% of 2.7 but 2.7 was over flowed and I had to cut the flow percent back for this PLA Plus filament.
Cura allows flow to be raised by up to 150% so I should be able to get up to 22g with 3.5.1.  I will try that and also see if I can measure parts of prints accurately to see if I can establish a "correct" flow level.  So, there is a work=round but I don't really understand what is going on.
If DjangoBlok tries 3.5.1 then it will be interesting to see whether he also gets a big increase in flow.  If it comes back to his 3.4.1 level then the drop to three quarters, for me, was between 2.7 and 3.4.1 but if he gets only about 75% then there is still a difference between 3.4.1 and 3.5.1 which would be good to try and understand.

DjangoBlok, 2018-10-17:
Yes, upgraded to 3.5.1 
switched material (and reselected original material)
The flow is great again!