Ultimaker/Cura #4583: [3.5.0] Supports are generated beneath the model under the "real" support structure

printingotb, 2018-10-15:
**Application Version**
3.5.0

**Printer**
Custom FDM

**Steps to Reproduce**
See attached project and pictures

**Actual Results**
Supports are generated beneath the model under the "real" support structure

**Expected results**
Same file and settings did not do this in 3.4.1

**Additional Information**
Cura 3.5.0:
[image]
Cura 3.4.1:
[image]

[supportbug.zip](https://github.com/Ultimaker/Cura/files/2480196/supportbug.zip)

diegopradogesto, 2018-10-16:
This seems to be a bug. The support is being generated from the top part and dropping down till the BP. I even tried to put a support blocker in the bottom part and the support is generated anyway.

[image]

gravesfamily, 2018-10-16:
Just to add on, I'm seeing the same thing in a different model that sliced fine in 3.4.1, and because of it, I cannot actually print the item I need because the support inside the tunnel fills it up, making the tunnel useless. :-)

Using this object: https://www.thingiverse.com/thing:2783459

The tunnel in the object ends up with a support in 3.5:
[image]

While 3.4.1 does NOT have the same support generated:
[image]

Note that there is actual model above the tunnel that will have support on top of it as necessary.
[image]
[image]

b14ckyy, 2018-10-26:
Confirmed. Recognized yesterday on my first slice on 3.5.1 too. 

If Printbed only, cura will generate printbed support for overhangs, even if parts of the model are underneath:

[image]

The big overhang on the back is ignored, this has a 45° angle. 

[image]

if support activated anywhere it looks like this:

[image]

if support blocker is used in the top part, the support beneath will disappear on that line:

[image]

dragoon88, 2018-11-14:
This bug has continued into version 3.6. It does not matter if there is support blocker on a lower overhang, or if the lower overhang is at a printable angle. The lower overhang angle in this picture is 33°. The other lower overhang has a support blocker.
[image]

Ghostkeeper, 2019-01-15:
It doesn't reproduce in Cura 4.0 any more:

[image]

This same model did exhibit the bug in 3.6.

Ghostkeeper, 2019-01-15:
Ah, yes, it looks like our internal ticket (CURA-5833) was already marked by our QA as fixed.

closed 2019-01-15 by Ghostkeeper