Ultimaker/Cura #4584: Feature request: Layer view: display absolute height corresponding to displayed layer

MB3Y, 2018-10-15:
**Application Version**
3.5.0
**Platform**
-
**Printer**
-
**Steps to Reproduce**
Activate Layer view and browse through the layers of a sliced model
**Actual Results**
Layer number is displayed, but height not. One can calculate the height, taking into account several parameters like initial layer height and layer height... but that is cumbersome, no? Besides, this method goes completely haywire when one uses adaptive layers.
**Expected results**
Display next to the number of the layer also the absolute height of the layer.
**Additional Information**
-

Thisismydigitalself, 2018-10-16:
#2536 was issued by me a year ago. few months later this issue got dismissed for the following reason:
"Our project manager removed this from our planning because it was not seen as important enough."

diegopradogesto, 2018-10-16:
I like the idea, but it's not in our current plan. I'll mark it as an improvement but I will close it since it's a duplicate of #2536. Let's see if somebody has time to implement it in the near future.

closed 2018-10-16 by diegopradogesto

MB3Y, 2018-10-16:
I don't agree with the duplication as the adaptive layer height feature didn't exist at that time - no?
Usecase example : I want a certain feature on my model to appear at a certain height (eg to match it with geometry in an existing physical assembly the object had to match with) - then I will want to sink the object in the buildplate - or scale it - until this feature is at a certain physical height.

diegopradogesto, 2018-10-16:
They way it should work is taking into account the current height of each layer, so it shouldn't matter if it's sliced with adaptive layer heights or not. That's why I consider it as a duplicate.