Ultimaker/Cura #4585: object name

53peterp, 2018-10-15:
There does not seem to be a way of identifying objects on the build plate by name. If I have say 4 items that look very similar it would be a great help if I could mouse over the object and the individual stl file name of that object would show.

diegopradogesto, 2018-10-16:
There is a feature called "Multi buildplate" that can do the trick for you. You can enable it in "Preferences > Configure Cura... > General". If you go to the bottom of this page you will see an Experimental feature that you can enable. That will add a list of models that you load in the buildplate.

Please be aware that this feature is still experimental.

53peterp, 2018-10-16:
Yes that's great I see the list of object stl file names thank you

closed 2018-10-16 by 53peterp