Ultimaker/Cura #4586: [3.5.0] Cura doesn't launch/opened after reboot/shutdown.

boomacoh, 2018-10-16:
**Application Version**

3.5.0
**Platform**

Windows 10
**Printer**

Tevo Tornado 2018
**Steps to Reproduce**

1. Reboot/Shutdown Windows
2. Reopen Cura 3.5.0.
**Actual Results**

No loading screen is shown when Cura is Launched.
A new instance of "cura.exe" is being added to task manager whenever launched.
Need to uninstall and install just to open cura.
**Expected results**

Even after rebooting or shutting windows down Cura should still be able to launch properly.
**Additional Information**

Cura 3.4.1 works perfectly even after a shutdown and reboot. I only noticed this bug or issue when I upgraded to 3.5.0
Already tried running it as administrator and still same results.

diegopradogesto, 2018-10-16:
Can you please send the log file to us? You can find it in your user data folder (in `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`. Let's see if there is something there that indicates the problem.

boomacoh, 2018-10-16:
[cura.log](https://github.com/Ultimaker/Cura/files/2481949/cura.log)
Here you go.
When I tried reopening it just gets stuck on last line
"2018-10-16 15:53:03,209 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [294]: Attempting to kill the engine process"

alekseisasin, 2018-10-16:
Hi @boomacoh , 
''....Attempting to kill the engine process" - It is fine, it is expected the message.
There is another error while saving the Gcode, Can try start Cura and close without doing any operations, reboot Windows and try to start Cura again.

Can you describe the steps you did to save the Gcode file, especially file location?

boomacoh, 2018-10-16:
> I see these errors:
> 
> ```
> 2018-10-16 11:13:22,338 - ERROR - [Thread-1] UM.Logger.logException [84]: Exception: An exception occurred while trying to send slice information
> 2018-10-16 11:13:22,347 - ERROR - [Thread-1] UM.Logger.logException [88]: Traceback (most recent call last):
> 2018-10-16 11:13:22,352 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 1254, in do_open
> 2018-10-16 11:13:22,355 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "http\client.py", line 1106, in request
> 2018-10-16 11:13:22,358 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "http\client.py", line 1151, in _send_request
> 2018-10-16 11:13:22,361 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "http\client.py", line 1102, in endheaders
> 2018-10-16 11:13:22,364 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "http\client.py", line 934, in _send_output
> 2018-10-16 11:13:22,366 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "http\client.py", line 877, in send
> 2018-10-16 11:13:22,368 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "http\client.py", line 1260, in connect
> 2018-10-16 11:13:22,370 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "ssl.py", line 377, in wrap_socket
> 2018-10-16 11:13:22,372 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "ssl.py", line 752, in __init__
> 2018-10-16 11:13:22,374 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "ssl.py", line 988, in do_handshake
> 2018-10-16 11:13:22,375 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "ssl.py", line 633, in do_handshake
> 2018-10-16 11:13:22,376 - ERROR - [Thread-1] UM.Logger.logException [88]: socket.timeout: _ssl.c:629: The handshake operation timed out
> 2018-10-16 11:13:22,378 - ERROR - [Thread-1] UM.Logger.logException [88]: 
> 2018-10-16 11:13:22,380 - ERROR - [Thread-1] UM.Logger.logException [88]: During handling of the above exception, another exception occurred:
> 2018-10-16 11:13:22,382 - ERROR - [Thread-1] UM.Logger.logException [88]: 
> 2018-10-16 11:13:22,384 - ERROR - [Thread-1] UM.Logger.logException [88]: Traceback (most recent call last):
> 2018-10-16 11:13:22,386 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "C:\Program Files\Ultimaker Cura 3.5\plugins\SliceInfoPlugin\SliceInfoJob.py", line 32, in run
> 2018-10-16 11:13:22,388 - ERROR - [Thread-1] UM.Logger.logException [88]:     f = urllib.request.urlopen(self._url, **kwoptions)
> 2018-10-16 11:13:22,390 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 163, in urlopen
> 2018-10-16 11:13:22,392 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 466, in open
> 2018-10-16 11:13:22,394 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 484, in _open
> 2018-10-16 11:13:22,395 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 444, in _call_chain
> 2018-10-16 11:13:22,397 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 1297, in https_open
> 2018-10-16 11:13:22,398 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "urllib\request.py", line 1256, in do_open
> 2018-10-16 11:13:22,400 - ERROR - [Thread-1] UM.Logger.logException [88]: urllib.error.URLError: <urlopen error _ssl.c:629: The handshake operation timed out>
> ```
> I'm not sure if it's the same issue as #4016. Can you please try what we suggest on this thread?

I just did what you suggested in that post. Disabling the Send (anonymous) print information. Which solve the problem. Even when I turn it back on. It is still able to launch properly now. But I'll still be on watch and check if it comes back. 

Thanks

nallath, 2018-10-16:
Note that these are errors that we catch and log, but are handled otherwise. The fact that they show up in the log doesn't
 that it's the actual cause of your issue.

diegopradogesto, 2018-10-16:
Yes, this is not the solution, but at least is a workaround for the user. It can be something with your network configuration, firewall, ... If you have more info, maybe we can go deeper.

Thanks for sharing your results.

boomacoh, 2018-10-21:
Unfortunately this didn't solve it entirely. I get to relaunch Cura even after closing it if has not yet been used that much after reboot. Which is why I thought your solution worked. Now after some hours of using it, it went back to the issue again. Even uninstall my python on my pc and also tried using 3.5.1 but nothing is working now. I just constantly uninstall and install just to use it. 
I'll be posting the new log again if in case it can help. Any other information you need to solve this just let me know. 

Thanks! 
[cura.log](https://github.com/Ultimaker/Cura/files/2498892/cura.log)

alekseisasin, 2018-10-22:
@boomacoh , I wonder, when you have the situation that you cannot start Cura. Can you open  the 
_C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\_  folder and check, do you have a   _cura.lock_ file there or not?

boomacoh, 2018-10-22:
[image]

There is no _cura.lock_ file in the directory you mentioned

Chucky911, 2018-10-23:
I had the same issue as you....here is how i solved it:
Open Windows Defender->Virus & Threat Protection settings->Exclusions-> add or remove exclusion

Click Add an Exclusion and select "Process"
Enter process name: Cura.exe

launch Cura

Let me know if this works for you !

boomacoh, 2018-10-28:
> I had the same issue as you....here is how i solved it:
> Open Windows Defender->Virus & Threat Protection settings->Exclusions-> add or remove exclusion
> 
> Click Add an Exclusion and select "Process"
> Enter process name: Cura.exe
> 
> launch Cura
> 
> Let me know if this works for you !

Sorry only now, but this is working for me. Had to test for a couple of days before I gave some feedback, so far I am able to open Cura after reboot or when I accidentaly close it.

Thank you!

diegopradogesto, 2019-09-11:
This issue seems to be solved by adding an exception to Windows Defender. However we are keeping an eye on this because it happens every so often and could be related with other libraries that are used in Cura. 

Thanks for the effort on tracking the issue.

closed 2019-09-11 by diegopradogesto