Ultimaker/Cura #4587: [3.5.0] Support roof created even when "Enable Support Interface" was not selected

ransildesign, 2018-10-16:
**Application Version**
Cura 3.5.0

**Platform**
Windows 10
**Printer**
Tevo Black Widow

**Steps to Reproduce**
Install Cura 3.5 without upgrading (I installed on a new computer)
create a new profile with "Enable Support Interface" selected from Preferences/Settings dialog box so it is an option in the right Print Setup panel but do not select "Enable Support Roof" or Enable Support Floor"
slice a part and create Gcode with "Enable Support Interface" NOT selected

See attached image for behavior of checkbox selection.

**Actual Results**
Even though "Enable Support Interface" was not selected a support roof was generated. 

**Expected results**
support roof should not be generated

**Additional Information**
I looked at the profile setting and noticed that "Enable Support Interface" was false but "Enable Support Roof" was true. I went back into the preferences/settings and checked the "Enable Support Roof"  and Enable Support Floor" options. I then could see that the "Enable Support Roof" was selected in the Print Setup even though "Enable Support Interface was still not checked in the Print Setup. As soon as I removed the check mark from "Enable Support Roof", the "Enable Support Interface" box grayed out and has a gray check mark. Even after deselecting "Enable Support Roof" and "Enable Support Floor" from the preferences/setting dialog, the Print Setup box shows "Enable Interface Support" as grayed out. I would expect this to override the selection of the sub-options. 

Upon further investigating is discovered that when "Enable Support Roof" option is selected the "Enable Support Interface" box is grayed out. This is not the case with "Enable Support Floor". Also, if all boxed are unchecked then I select the "Enable Support Roof" the "Enable Support Interface" box is still unchecked but the slicer generated the roof support layers. I would expect the status of "Enable Support Interface" to over ride either of the sub-options.  
The function of "Compensate Wall Overlaps" and it's sub-options should behave in the same way but doesn't. If all boxes are checked and the parent box is de-selected the children boxes are also de-selected. Oddly though, if all boxes are de-selected it is possible to select a child box without the parent box automatically being selected. I don't know how this would affect the execution of this option during slicing but it is possible that it could lead to similar confusion.
[image]
[image]

diegopradogesto, 2018-10-17:
This is because in the profiles for Tevo Black Widow, the setting `support_roof_enable` is set to true. So Cura works as expected but maybe we can remove these lines in the profiles. Since we don't use that printer, we don't know if this is right or wrong, so we trust our users.

diegopradogesto, 2018-10-17:
I've just changed it since I think it makes sense.

closed 2018-10-17 by diegopradogesto

AndreKR, 2020-02-05:
I see the same problem with the Anycubic i3 Mega.

commit e9e95b85 (2018-10-17): Remove support_roof_enable to True in Tevo Black Widow quality profiles,
since the support interface is already enabled.

Fixes #4587.