Ultimaker/Cura #4591: Software Bug for 3.5.1

kornshadow097, 2018-10-16:
**Application Version**
3.5.1

**Platform**
windows 10 64bit

**Printer**
tevo tarantula

**Steps to Reproduce**
 Anytime I select a different nozzle size profile it crashes. i was on 0.6mm and it was fine, i switched to my 0.8mm and it froze but came back and when i switched to 1.0 the program crashed 5x. so then i tried to go back to 0.6mm and it just keeps crashing everytime i open the program and try to revert to my nozzle size

**Actual Results**
it crashes

**Expected results**

**Additional Information**

nallath, 2018-10-16:
Could you also post your logs?

Abraiator, 2018-10-17:
Application Version
3.5.1

Platform
windows 10 64bit

Printer
Framebox Prusa Clone
Foldarap

Similar thing here. Both with the 3.5.0 and the 3.5.1 versions. 

Every time I change the printer in use (in the top right corner menu) Cura crashes.

My printer profiles are inherited from previous Cura versions and always worked flawlessly before.

nallath, 2018-10-18:
Same as before; I need logs to diagnose the problem. If no-one provides them, there is nothing I can do to resolve the issue.

Abraiator, 2018-10-18:
Ok, I'm attaching a log (in order to keep it small I deleted the old one, opened Cura, tried to change the printer on the right top menu and... voilá! Cura crashed.

Thanks.
[cura.log](https://github.com/Ultimaker/Cura/files/2492664/cura.log)

ZerialLim, 2018-10-21:
Having the same problem. Attaching my own log.
[cura.log](https://github.com/Ultimaker/Cura/files/2499379/cura.log)

ZerialLim, 2018-10-21:
0. According to the log file, there must be something wrong with the extruder config file.
1.  I deleted all extruder cfg files from /definition_changes
2. Run Cura
3. Configuration Error message pops up, saying fdmextruder settings and robo3dR1 extruder settings are corrupted (yeah that's the files I removed)
4. Printer change works fine.

ZerialLim, 2018-11-02:
problem continues after reboot.

nallath, 2018-11-06:
I think the issue has been fixed for the 3.6 final release. Could you try again once it's released?

Ghostkeeper, 2019-02-19:
This was fixed in 3.6 stable.

closed 2019-02-19 by Ghostkeeper