Ultimaker/Cura #4595: Layered view no longer working

thomaskilian, 2018-10-17:
**Application Version**
3.5.1

**Platform**
Mac

**Printer**
DIY

**Steps to Reproduce**
Render any model and choose the 3d view. The German name is Schichtenansicht, which is probably Layered View.
Well, aside the fact that moving the layer slide is now slow as molasses it does not show any layer at all. Compared to the previous version I had (was it 3.4?) where I could smoothly scroll through the single layers it's now a how-can-I-effectively-waste-CPU-time.

Ghostkeeper, 2018-10-18:
Oh, from here it seems to work fine, also on MacOS. Was it a particularly heavy model? Lots of g-code? Perhaps the layer view data is still processing?

thomaskilian, 2018-10-18:
Well, no. It's not complex and as said it worked smoothly before. I skipped the version before 3.5.0 which is also unusable. Here's the .stl file:

[ControllerCase.gcode.zip](https://github.com/Ultimaker/Cura/files/2493629/ControllerCase.gcode.zip)

thomaskilian, 2018-10-20:
This is what it looks like now:
[image]

And that's the 3.4.1 output:
[image]

Except that 3.5.1 is now slower than a snail it doesn't have any valuable improvements (for me). So I'm just reverting back.

diegopradogesto, 2018-11-08:
Maybe related with the compatibility mode in the layer view? Does it happen when loading a gcode or after slicing a STL? Can you please share a project file with us? (In 3.4, File > Save Project...)

thomaskilian, 2018-11-08:
When I load the G-code it's different as the body is shown completely blue. But the layer can't be distinguished anyway. Also dragging the layer slider is awfully slow in any case. No comparison to 3.4.1. I'm working on OSX High Sierra. Here's a sample stl/gcode:

[Archiv.zip](https://github.com/Ultimaker/Cura/files/2563520/Archiv.zip)

diegopradogesto, 2018-11-19:
I can't do anything to reproduce your environment with STL and Gcode, that's why I asked for a project file (In 3.4 is `File -> Save Project...`, in 3.5 is `File -> Save...`). Please attach it to investigate more.

thomaskilian, 2018-11-20:
Well, no. I just can state that 3.4 works fluently when sliding through layers and from 3.5 on it's anything else but fluent. It's more of the viscosity of tar. Will try on my other (older) Mac at home.

Ghostkeeper, 2020-04-21:
You can change the colours from "material color" to "line type" in the drop down in the top-centre of the screen. That's why it appears all blue for you.

If performance is an issue, you can revert to the old behaviour by enabling the compatibility mode for layer view in the preferences.

closed 2020-04-21 by Ghostkeeper