Ultimaker/Cura #4601: print job stopps after  some hours 3.5.1

robschii, 2018-10-17:
**Application Version**
3.5.1
**Platform**
win10

**Printer**
mkc4.1 marlin last stable 1.1.9 marlin  

**Actual Results**
printer stops in the middle of the printjob,  no axis is moving anymore,   i had to continue manaully the printjob.  but im not in the room so my steppmotors are deactivate after 10 minutes

some days ago my z driver was broken by this event

rburema, 2018-10-18:
Hey @robschii, This seems to be similar to #3994 (or at least the fallout later in the discussion). If so, this will be fixed by PR #4487, and will hopefully be in the next release. It seems that the last release we made that didn't suffer from this might've been 3.3.1. cheers, Remco.

diegopradogesto, 2018-10-19:
I do think this is a duplicate of #3994.

closed 2018-10-19 by diegopradogesto

robschii, 2018-10-21:
i print from sd card

rburema, 2018-10-26:
Hi @robschii, That decreases the likelihood that it has to do with Cura; it might be a firmware or some other printer-related issue. Does this occur with earlier versions or other slicers as well, or is it just 3.5.1? cheers, Remco.