Ultimaker/Cura #4603: Scale To Specific Percentage

matt7c, 2018-10-17:
**Application Version**
3.5.0 
**Platform**
MacBook Pro 

-Cura/gui seems really slow but not the main issue! (Maybe its just my 2011 platform)
-Is it possible to add icons in the menus showing the different patterns (e.g. ZigZag, Concentric, ...) for faster reference. Some are obvious, but others not so much!
-Also option to switch between various units (e.g. inches) or convert models upon import or scale models to a specific % upon import (vs. Auto Scaling).

matt7c, 2018-10-18:
as per the build height, my gantry was set to 30mm, adjusted to 1000m and the problem seams to be solved. The build height doesn't automatically adjust . (i always Print One At A Time btw)

rburema, 2018-10-18:
Hi @matt7c, Please separate these into multiple issues. A mixed bag of feature requests and potential bugs isn't really workable, sorry. cheers, Remco.

Ghostkeeper, 2018-12-24:
> Cura/gui seems really slow but not the main issue! (Maybe its just my 2011 platform)

That's an ongoing effort. 4.0 made some improvements in this respect when changing profiles.

> Is it possible to add icons in the menus showing the different patterns (e.g. ZigZag, Concentric, ...) for faster reference. Some are obvious, but others not so much!

Yes. Good idea.

> Also option to switch between various units (e.g. inches) or convert models upon import or scale models to a specific % upon import (vs. Auto Scaling).

Try the Barbarian Plug-in in the Marketplace. It causes Cura to load objects that are saved in inches (i.e. scales them to 2540% upon loading).

> Build Height is automatically adjusted after import of first part and can't print multiple parts of different height as in previous versions of CURA, even when spaced out appropriately. Creality-10S 300x300x400mm

I think you've set Print Sequence to one-at-a-time then. Please revert it to All-at-once to use the full height of your printer.

closed 2020-06-20 by no-response[bot]