Ultimaker/Cura #4604: printing by zero?

teknomancer2008, 2018-10-18:
hi, ive been trying to print in 3d for a while now. finally im making some head way, your software seems to be the best for me, and my ctc prusa i3 clone. however, there is one thing that occurred today that amazed me. the printer was told by your software to print at zero temperatureof both the heat bed and the nozzle! odd. i have no idea why, but thought it best to let you know whats going on. only way to stop it is to force a pre heat, then restart the print. thanks.

smartavionics, 2018-10-18:
Hi, please save the project (.3mf) file, zip it up and attach to this issue so we can look at the settings. It would also be useful to see the gcode file as well so please put that in the zip also. Without that info, it's not possible to work on your issue. Thanks.

diegopradogesto, 2018-10-19:
We need you to fill out the template when you open an issue. This is very important for us because we need to know the version you're using, printer type, and some other information. Also, please provide the project file, as indicated by @smartavionics .

nallath, 2018-11-20:
Closing due to inactivity.

closed 2018-11-20 by nallath