Ultimaker/Cura #4605: M107 is deprecated

thomaskilian, 2018-10-18:
According to [RepRap](https://reprap.org/wiki/G-code#M107:_Fan_Off) M107 is deprecated. However, Cura sends it initially at layer 0:

> ;----------------
;LAYER:0
M107
G0 F3600 X51.92 Y82.155 Z0.2 
...

stelgenhof, 2018-10-18:
That is not correct. The documentation says "Deprecated in **Teacup** firmware and in **RepRapFirmware**.". Firmwares like Marlin for example still supports this GCode command (See: 
http://marlinfw.org/docs/gcode/M107.htm)

thomaskilian, 2018-10-18:
Unfortunately the Teacup flavor is not available :-/

thomaskilian, 2018-10-18:
Probably I have to do that in my Teacup post-processor. Would just need to remove that very line from the record. Well....

stelgenhof, 2018-10-18:
In that case I would add the M106 command in the start GCode section :)

thomaskilian, 2018-10-18:
That does not stop Cura to send M107. However, as long as I can post-process the code it's probably just fine. You may close this issue.

stelgenhof, 2018-10-18:
Correct it will still send, but shouldn't be an issue when you put the M106 S0 command in.
(I can't close this issue, only you or the maintainer(s) of this repository can).

Cheers! Sacha

closed 2018-10-18 by rburema