Ultimaker/Cura #4606: Suport on top surfaces an

tebz, 2018-10-18:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

rburema, 2018-10-18:
Hi @tebz, I'm not sure what you're trying to say here? Please finish the title, fill out the form, and possibly attach a model and/or screenshots. cheers, Remco.

rburema, 2018-10-26:
Hm, apearantly github doesn't notify a commenter when a report is edited only. Anyway, we're getting similar reports (see also #4658 and #4653 ), so we'll probably have a look at it.

rburema, 2018-10-26:
Possible relation to #4583 as well?

Ghostkeeper, 2018-10-29:
It's generating this support because your model is not watertight:

[image]

This introduces tiny (~20μm) pieces of overhang which are expanded by 3mm due to the Support Horizontal Expansion setting.

alekseisasin, 2018-10-29:
Devs, see CURA-5868.

tebz, 2018-10-30:
Would it not be logical to ignore overhang smaller than, maybe, upp to half the printing width ?

rburema, 2018-10-31:
Hi @tebz,
We enforce water-tightness in order to do these sorts of calculations (overhang-angle, etc.) in a timely manner, so slicing time (not to be confused with printing time) does not become prohibitive. Also it isn't really well defined what the 'outside' and 'inside' of an object is if water-tightness is violated. That means that all bets are off if a model doesn't conform.
You could argue that we should just disable all support smaller than a certain size, but then things that (locally) look like, say, an iverted pyramid woudn't have support printed any more.
cheers, Remco.

Ghostkeeper, 2019-01-22:
Removing all small areas would invalidate our option for support towers. Plenty of prints need to support things that are very slim: Pins pointing downwards, wings with wingtips, clips with sharp edges, etc.

You can set the Support Minimum Area setting higher in your case, which would fix the issue. I think that is a good workaround, and more or less what you were looking for anyway, but it's optional.

closed 2019-01-22 by Ghostkeeper