Ultimaker/Cura #4610: Scaling Issues

StarsnStripes, 2018-10-18:
Hey there,

I got some issues while scaling my .stl file. The assignment of x,y,z to their percentage doesn't seems right.
You will find some screenshots and the project in my attechment.

Kind regards!

**Application Version**
Cura 3.5.1

**Platform**
Windows 7

**Printer**
Other: ANET A8

**Additional Information**
original:
[image]

scaled:
[image]

[scaling_issues.zip](https://github.com/Ultimaker/Cura/files/2492363/scaling_issues.zip)

rburema, 2018-10-18:
Hi @StarsnStripes, While not quite the same behaviour, there's something strange going on in the current master as well with your model when I tried those steps. Will discuss it with the team. cheers, Remco.

diegopradogesto, 2018-10-19:
This is a duplicate of #3625.

closed 2018-10-19 by diegopradogesto