Ultimaker/Cura #4611: [3.5.1] GCODE Variables / Macro for RAND()

mrambossek, 2018-10-18:
**Application Version**
3.5.1 (or any, really)

**Platform**
any

**Printer**
any, doesnt matter

**Steps to Reproduce**
not applicable

**Actual Results**
not applicable

**Expected results**
not applicable

**Additional Information**

this is a feature request.
it would be awesome to have a GCODE macro that allows you to insert a RAND() value and/or apply it to variables for GCODE generation.

the particular use case i am thinking about is..
i am usually re-using the blue tape on my alu hotbed for 5-15 prints, depending on damage from removing too-sticky prints. since my start gcode includes 2 "wipe lines" close to/alongside the left border, there are specific locations on the tape that will get "used" EVERY single print, thus wearing down faster.

it would be nice to be able to have something like a RAND() value stored in variable. my current START code looks like this:

```
[...]
G1 X0.1 Y20 Z0.4 F5000.0 ; Move to start position
G1 X0.1 Y200.0 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 F5000.0 ; Move to side a little
G1 X0.4 Y20 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
[...]
```

and it would be nice to be able to do something like this:
```
[...]
{my_x=0.1 + rand() * 20} ; create random xpos between 0.1 and 20.1
G1 X{$my_x} Y20 Z0.4 F5000.0 ; Move to start position
G1 X{$my_x} Y200.0 F1500.0 E15 ; Draw the first line
{my_x=$my_x+0.3}
G1 X{$my_x} Y200.0 F5000.0 ; Move to side a little
G1 X{$my_x} Y20 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
[...]
```

i am guessing the RAND() part would be easy; the variable handling / eval()ing of math would be the tougher stuff... ?

Ghostkeeper, 2018-12-24:
I think it'd be a pretty arcane option if we would add it. You're also asking for the possibility to define variables in your start g-code and compute sums like `+0.3`, making it a lot more powerful. It's a lot of work though to implement a system like that.

A pull request would be welcome since it's not in the way for anyone, but this won't have much priority for us.

mrambossek, 2023-06-29:
its now 5 years later, i dont have a 3d printer anymore :) closing.

closed 2023-06-29 by mrambossek