Ultimaker/Cura #4612: [Cura 3.5.x] Reverts to Chinese language on Wanhao/Monoprice

itzameongithub, 2018-10-19:
**Application Version**
3.5 and 3.5.1

**Platform**
Windows x64

**Printer**
Monoprice Maker Select Plus (Wanhao Duplicator i3 Plus)

**Steps to Reproduce**
Print an object

**Actual Results**
Printer menu switches to Chinese at the end of the print.

**Expected results**
Printer stays at English.

**Additional Information**
This does not happen with every print, but it's only started since 3.5 after I changed Cura to use the Wanhao Duplicator i3 Plus as my printer since that's what the Monoprice Maker Select is. I understand that Wanhao is Chinese, but it shouldn't be changing my display to Chinese. This never occurred before 3.5 came out, but it's happening about 1/3 of the time.

LipuFei, 2018-10-19:
Hi @itzameongithub Does this also happen if you use 3.4.1 and perform the same steps?

fieldOfView, 2018-10-19:
There's hardly any change between 3.4 and 3.5 when it comes to USB printing. It is more interesting to know if it also happens with Cura 3.3. 

The root cause of #3994 is that in 3.4 and 3.5, Cura can cause the serial buffer to overflow on the printer. I have no idea what Wanhao did with the firmware, but buffer overflows can have strange sideeffects.

itzameongithub, 2018-10-19:
I'm trying to isolate any commonality or consistency to it. It happened again last night but on an 11 hour print, so I really don't want to have to run that again. I wasn't sure if there was some kind of errant gcode that might trigger it. Also, this isn't USB printing. It's printing via SD card. I'm not ruling out firmware issues. Maybe I'll use this as my excuse to install ADVi3++, but it seems odd that this never happened until I upgraded to Cura 3.5.  It didn't occur with any previous versions.

fieldOfView, 2018-10-19:
What version of Cura did you upgrade from?

fieldOfView, 2018-10-19:
Do you have your printer connected through USB?

itzameongithub, 2018-10-19:
SD card.

fieldOfView, 2018-10-19:
That's not what I asked; is the usb port connected? You can print from SD, while having the usb cable connected to a computer.

itzameongithub, 2018-10-19:
Nope. I've never used the USB connector on it.

fieldOfView, 2018-10-19:
Ok, then the #3994 issue does not apply to you, and is not the cause of your issue.

fieldOfView, 2018-10-19:
AFAIK, there is no gcode to change the display language. That makes this a firmware problem.

itzameongithub, 2018-10-21:
It happened again on several prints, but it looks like it really is a matter of coincidence that it was after I upgraded to Cura 3.5. I did some further digging and (finally) found a single thread on Reddit from a year ago for this exact issue and Monoprice said that it's the control board possibly going bad, Sorry for wasting your time, but thanks for at least helping to focus the search for an answer a bit better.

closed 2018-10-21 by itzameongithub

fieldOfView, 2018-10-21:
Thanks. Perhaps you can post the Reddit thread here, so if it happens again to someone else we can refer them.

itzameongithub, 2018-10-21:
Sure thing. There's not much to it, though.

https://www.reddit.com/r/3Dprinting/comments/605k32/monoprice_maker_select_plus_i3_plus_wierd/

itzameongithub, 2018-10-22:
FYI.... this might not be a done deal. The first thing that Monoprice asked was if I recently upgraded my Cura, which I obviously did.

> This may not actually be the control board. Have you changed versions of CURA or are you using a different slicer than you used to recently? 

> We have noticed that there is recurrence of this when customer update the slicer and when they revert back to what they were using before. This has only been tested by customers at this point and we have been unable to replicate the issue or it's fix in our facilities. 

So I'm going to be doing some testing over the next week or so to see if I can isolate it. Hopefully, it's nothing more than changing the Wanhao configuration to switch some gcode entries around.

itzameongithub, 2018-10-26:
At this point I think I'm going to let this one go. Unfortunately, I can't get any consistency or find any specific trigger. I printed off five different object three times each: once using my Cura 16.03 that came with the printer, once with 3.5.1 and the Wanhao setting, once with 3.5.1 with the Prusa i3 setting, since this is basically a Prusa i3 clone.

None of the 16.03 prints had an issue, 3 of the Wanhao prints switched to Chinese, and 1 of the Prusa prints switched. Size didn't matter because the one print where both the Wanhao and Prusa setting switched to Chinese was a set of small cylinders. One of my larger prints didn't encounter this problem on any of them.

And, let's face it, this is a very minor cosmetic issue. It doesn't appear to be affecting the printer itself. So even if I could find some correlation, it's just not that big of a deal. So I think at this point I'm just going to let it go. Thanks for giving it some attention anyway.