Ultimaker/Cura #4617: How to create a quality profile for a printer with multiple extruders???

Passador99, 2018-10-19:
I'm trying to create a quality profile for a new printer wich is not present in cura by default. i cannot figure out how. I've the .curaprofile for the custom printer, I created all the .def.json files and they work properly when I add the new machine. I cannot see the inst.cfg profile that i created for each extruder.

rburema, 2018-10-19:
Hi @Passador99, Since this is not a bug or a feature request, please use the [forums](https://community.ultimaker.com/). cheers, Remco.

closed 2018-10-19 by rburema

LipuFei, 2018-10-19:
Hi @Passador99 , please take a look at our Wiki page. We have this page which describes this: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura