Ultimaker/Cura #4619: Filament diameter settings dont work

hagan66, 2018-10-19:
Entering different filament diameters in materials settings makes no difference in extrusion. I also tried changing diameter in the Machine settings. No effect. I have made a complete re-install of Cura resetting everything. I also wonder where the right place is to enter the measured filament diameter now since you removed it from the material settings in the sidebar.

**Application Version**
3.5.1

**Platform**
Mac

**Printer**
Prusa i3 mk2s

**Steps to Reproduce**
Print single wall cube, measure wall thickness. Change diameter settings, print again and measure.

TheNitek, 2018-10-20:
Probably a duplicate of #4582

diegopradogesto, 2018-10-24:
Yes, I think it's the same. Fixed in the upcoming Cura 3.6.

closed 2018-10-24 by diegopradogesto