Ultimaker/Cura #4621: PauseAtHeight.py should probably ask for pause command rather than assume M0

TheRealTachyon, 2018-10-20:
**Application Version**
3.4-3.5.1

**Platform**
Linux x86_64 (running from appimage)

**Printer**
Monoprice Select Mini V2

**Steps to Reproduce**
slice an object
run Extenstions -> Post Processing -> modify G-Code
Choose Pause at Height from Add Acript
Chose desired options
Close

**Actual Results**
Script adds pause code snippet to g-code output as expected
When run, printer does not pause as expected 

**Expected results**
The printer should pause and move the print head to the chosen park location then wait for user input to resume

**Additional Information**
The problem (apparently) comes from line 290 of the current PauseAtHeight.py script which reads:
`prepend_gcode += self.putValue(M = 0) + ";Do the actual pause\n"`

This inserts the following in the actual g-code
`M0;Do the actual pause`

The correct g-code line should be:
`M600;Do the actual pause`

Which works as expected.

IMO, the input fields for this script should include one more line under "Standby Temperature" similar to
"printer's pause g-code (eg M0)"
The input should be checked to see if it fits the format "Mnnn" and then copied into a variable and used as the other input lines. Replacing the static line in line 290 as well with a insertion of the user input for the pause command

[image]

diegopradogesto, 2018-11-08:
That of course will depend on the firmware that you use. If I remember correctly, your printer has a version of Marlin, so in the documentation of Marlin, there are two different M-codes for different purposes:
**M0 (Unconditional stop)**: http://marlinfw.org/docs/gcode/M000-M001.html
**M600 (Filament change)**: http://marlinfw.org/docs/gcode/M600.html

If this doesn't work in your printer is because the firmware doesn't have the M0 code implemented, but that doesn't mean that we need to change it.

I suggest to use the filament change script that adds a M600 (https://github.com/Ultimaker/Cura/blob/3ab8568a8789cc254e7799fbd4c8d63155772686/plugins/PostProcessingPlugin/scripts/FilamentChange.py#L59). Try if this works. If not, I remember that somebody in the community was trying to unify all the "Pause at height" scripts into one, but I can't remember what is the status.

nallath, 2019-03-01:
Closing due to inactivity

closed 2019-03-01 by nallath

Ghostkeeper, 2019-03-01:
You could also use a search-and-replace script to change the `M0` into something else.