Ultimaker/Cura #4622: Unable to load files when uart-usb converter is attached

lujji, 2018-10-20:
I am unable to open any stl or project files on my machine until I disconnect an ft232 uart-usb converter, which I have constantly plugged in.

**Application Version**
3.5.x

**Platform**
Linux (Debian 10)

Ghostkeeper, 2018-12-27:
We're unable to test this without the ft232 uart-usb converter, right? We can't begin to debug this until we know what about that converter is preventing you from opening files.

lujji, 2018-12-27:
Right, although this [chip](https://www.ebay.com/sch/i.html?_from=R40&_trksid=p2334524.m570.l1313.TR12.TRC2.A0.H1.Xft232.TRS0&_nkw=ft232&_sacat=0&LH_TitleDesc=0&_osacat=0&_odkw=ftdi++ft232&LH_TitleDesc=0) is used in many 3d printer boards, microcontroller dev-boards, etc. I suspect the problem occurs when Cura scans serial ports, tries to detect a printer and gets no response.

closed 2020-06-20 by no-response[bot]