Ultimaker/Cura #4624: Line Width Setting gets lost

TheNitek, 2018-10-20:
**Application Version**
3.5.1, also happend in 3.5.0 and if I remember correctly also in 3.4

**Platform**
Windows 10 64bit

**Steps to Reproduce**
Set a Line Width that's not the default, safe to profile

**Actual Results**
Sometimes (didn't find a pattern yet) it reset to default/calculated line width (and doesn't show as changed even though the profile has not been saved again) - I did NOT change the noozle size!

**Expected results**
Line Width stays as set/stored

[ALP_sp320-mount.zip](https://github.com/Ultimaker/Cura/files/2498473/ALP_sp320-mount.zip)

TheNitek, 2018-11-04:
I think it happens when switch to another printer profile, exit Cura, open it again and switch back to the original profile. Z-offset is one saved in the profile and not changed - It just keeps jumping back to 0.

nallath, 2018-11-06:
As far as I can see it correctly stored the line_width (0.35) in your quality_changes profile. 

Could you, the next time it happens, share your project file again?

TheNitek, 2018-11-06:
Just happend again:
I changed the line_width to 0.35 and safed it, then I switched profiles a couple of times and restarted cura. First I noticed that the "support line width" wasn't visible anymore (which it was before). Then I switched into another profile and safed this other profile, restarted cura und switched back to the original profile. Now it says 0.4 again (but "support line width" is visible again?!)
[Untitled.zip](https://github.com/Ultimaker/Cura/files/2554714/Untitled.zip)

TheNitek, 2018-11-06:
It just happend again Steps to reproduce:
Change line width to 0.35
Restart Cura once: "Support line width" settings becomes invisible
Restart Cura again: "Support line width" appears again
Restart Cura again: line width is back to 0.4

[Untitled.zip](https://github.com/Ultimaker/Cura/files/2554737/Untitled.zip)

nallath, 2018-11-08:
Uh... Where did you get your machine definition from? It's full of mistakes. So full, that I can't even load the project.

TheNitek, 2018-11-08:
I used this https://www.thingiverse.com/thing:2367365 and migrated it through a few cura version over time.
[stderr.zip](https://github.com/Ultimaker/Cura/files/2563126/stderr.zip)

closed 2020-02-13 by TheNitek