Ultimaker/Cura #4625: Anet A8

DavidMnDenver, 2018-10-21:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

DavidMnDenver, 2018-10-21:
I am using an Anet A8 on MacOS 10.14.1 over Cura 3.5.1. Repetier, on the same station, sees and controls the printed but Cura can't find it. I tried loading the newest firmware I could find and have changed the timeout to 5 - nothing.

Any ideas?

honza-kasik, 2018-10-26:
Are you trying to connected with both Cura and Repetier at the same time?

DavidMnDenver, 2018-10-26:
Sequentially.

Ghostkeeper, 2018-12-27:
How do I go about debugging this without having the Anet A8 printer available? What is it about A8 that prevents Cura from finding the printer?

closed 2020-06-20 by no-response[bot]